bblanchon/ArduinoJson #985: utf8 deserialization NotSupported ?

Vebryn, 2019-05-09:
Hello,

First of all, thank you for this awesome library!

I'm using arduinojson to parse opendata from city bus transportation. Some lines contains non ascii chars like "éè" represented by "\u00e9".

Deserialization succeed in parsing json from start of json to concerned char.

I'm using ESP8266 with Arduino IDE 3.

Are chars like "éè" supported ?

Thank you,
Best regards.

Code
```
  const size_t capacity = 3*JSON_ARRAY_SIZE(1) + 8*JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(4) + 4*JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(3) + 4*JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(6) + 4*JSON_OBJECT_SIZE(14) + 2340;
  String payload = curl("https://data.explore.star.fr/api/records/1.0/search/?dataset=tco-bus-circulation-passages-tr&rows=4&sort=depart&geofilter.distance=48.12207%2C-1.63874%2C200&apikey=aaaaaaaaaaaaaaaaaaaaaaaaaa", fingerprint);

  // Allocate the JSON document
  DynamicJsonDocument doc = parseJson(payload, capacity);

  // Extract values
  Serial.println("");
  //Serial.println(F("Response:"));

  for (int i = 0; i < 4; i++){
    Serial.print("nomcourtligne: ");
    Serial.print(doc["records"][i]["fields"]["nomcourtligne"].as<char*>());
    Serial.print(" destination: ");
    Serial.print(doc["records"][i]["fields"]["destination"].as<char*>());
    Serial.print(" arrivee: ");
    Serial.println(doc["records"][i]["fields"]["arrivee"].as<char*>());
  }
```

Debug
```
[HTTPS] GET... code: 200
{"nhits": 124, "parameters": {"dataset": ["tco-bus-circulation-passages-tr"], "timezone": "UTC", "rows": 4, "sort": ["depart"], "format": "json", "geofilter.distance": ["48.12207,-1.63874,200"]}, "records": [{"datasetid": "tco-bus-circulation-passages-tr", "recordid": "ed7fdce67d34356c8aa7acb334cf2ee9b16751f5", "fields": {"nomcourtligne": "C4", "depart": "2019-05-09T09:17:00+00:00", "departtheorique": "2019-05-09T09:17:00+00:00", "arriveetheorique": "2019-05-09T09:17:00+00:00", "idarret": "1313", "destination": "ZA Saint-Sulpice", "nomarret": "Beaulieu Restau U", "precision": "Applicable", "idcourse": "268436246", "arrivee": "2019-05-09T09:17:00+00:00", "sens": 1, "coordonnees": [48.121826, -1.638248], "idligne": "0004", "dist": "45"}, "geometry": {"type": "Point", "coordinates": [-1.638248, 48.121826]}, "record_timestamp": "2019-05-09T06:19:00+00:00"}, {"datasetid": "tco-bus-circulation-passages-tr", "recordid": "07b88ca6cc80ac72e86207568f7fdb073b27b37d", "fields": {"nomcourtligne": "14", "depart": "2019-05-09T09:17:00+00:00", "departtheorique": "2019-05-09T09:17:00+00:00", "arriveetheorique": "2019-05-09T09:17:00+00:00", "idarret": "1177", "destination": "Roazhon Park", "nomarret": "Beaulieu Restau U", "precision": "Applicable", "idcourse": "268437838", "arrivee": "2019-05-09T09:17:00+00:00", "sens": 0, "coordonnees": [48.122317, -1.639592], "idligne": "0014", "dist": "68"}, "geometry": {"type": "Point", "coordinates": [-1.639592, 48.122317]}, "record_timestamp": "2019-05-09T06:19:00+00:00"}, {"datasetid": "tco-bus-circulation-passages-tr", "recordid": "e9c9a9e0ab67709a70ffed6c1acc92686e7d6f0d", "fields": {"nomcourtligne": "C4", "depart": "2019-05-09T09:15:00+00:00", "departtheorique": "2019-05-09T09:15:00+00:00", "arriveetheorique": "2019-05-09T09:15:00+00:00", "idarret": "1292", "destination": "Grand Quartier", "nomarret": "Beaulieu Restau U", "precision": "Applicable", "idcourse": "268436416", "arrivee": "2019-05-09T09:15:00+00:00", "sens": 0, "coordonnees": [48.121437, -1.638194], "idligne": "0004", "dist": "81"}, "geometry": {"type": "Point", "coordinates": [-1.638194, 48.121437]}, "record_timestamp": "2019-05-09T06:19:00+00:00"}, {"datasetid": "tco-bus-circulation-passages-tr", "recordid": "3623260bc3a1ab83ce25c3dc20d8f2fd3f37eb32", "fields": {"nomcourtligne": "64", "depart": "2019-05-09T09:13:00+00:00", "departtheorique": "2019-05-09T09:13:00+00:00", "arriveetheorique": "2019-05-09T09:13:00+00:00", "idarret": "1203", "destination": "Acign\u00e9", "nomarret": "Beaulieu Restau U", "precision": "Applicable", "idcourse": "268440250", "arrivee": "2019-05-09T09:13:00+00:00", "sens": 1, "coordonnees": [48.122308, -1.639183], "idligne": "0064", "dist": "42"}, "geometry": {"type": "Point", "coordinates": [-1.639183, 48.122308]}, "record_timestamp": "2019-05-09T06:19:00+00:00"}]}
deserializeJson() failed: NotSupported

nomcourtligne: C4 destination: ZA Saint-Sulpice arrivee: 2019-05-09T09:17:00+00:00
nomcourtligne: 14 destination: Roazhon Park arrivee: 2019-05-09T09:17:00+00:00
nomcourtligne: C4 destination: Grand Quartier arrivee: 2019-05-09T09:15:00+00:00
nomcourtligne: 64 destination:  arrivee: 
```

Vebryn, 2019-05-09:
I just read #204. Will https://arduinojson.org/v6/api/config/decode_unicode/?utm_source=github&utm_medium=issues solve my issue ?

bblanchon, 2019-05-10:
Hi @Vebryn,

Yes, it should solve your problem.
I just added that to the FAQ:

* [Why does ArduinoJson return NotSupported?](https://arduinojson.org/v6/faq/why-does-arduinojson-return-notsupported/)

Best Regards,
Benoit

closed 2019-05-10 by bblanchon