bblanchon/ArduinoJson #987: 'ArduinoJson::JsonObject' has no member named 'prettyPrintTo'

Slyke, 2019-05-10:
Please provide all the relevant information:
* short description of the problem:
Upgraded to v5 to v6. Can no longer pretty print.

* target platform:
Arduino (ESP8266)

* compiler model and version:
I think it's `xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9`

* MVCE (https://stackoverflow.com/help/mcve):
I extended the I2C example to post results to an IP address. This is the function that has the offending code
```
void ICACHE_FLASH_ATTR txI2CData(int i2cAddresses[], int i2cAdressSize, int i2cAddressesErrors[], int i2cAdressErrorSize) {
  if ((currentReading.runMode & 8) == 0) {
    if ((WiFi.status() == WL_CONNECTED)) {
      WiFiClient client;
      HTTPClient http;

      if (http.begin(client, runTimeVariables.i2cPostUrl.c_str())) {
        http.addHeader(F("Content-Type"), F("application/json"));

        StaticJsonDocument<512> JSONbuffer;
        char JSONmessageBuffer[512];

        JsonObject JSONencoder = JSONbuffer.to<JsonObject>();
        JsonArray foundDevicesArray = JSONbuffer.to<JsonArray>();
        JsonArray errorDevicesArray = JSONbuffer.to<JsonArray>();
        
        JSONencoder["scanResults"] = JSONbuffer.to<JsonObject>();
        
        for (int i = 0; i < i2cAdressSize ; i++) {
          foundDevicesArray.add(i2cAddresses[i]);
        }

        for (int i = 0; i < i2cAdressErrorSize ; i++) {
          errorDevicesArray.add(i2cAddressesErrors[i]);
        }

        JSONencoder["scanResults"]["foundDevices"] = foundDevicesArray;
        JSONencoder["scanResults"]["errorDevices"] = errorDevicesArray;

        JSONencoder.prettyPrintTo(JSONmessageBuffer, sizeof(JSONmessageBuffer)); // <--- This one fails.
        
        lastTxResponse = http.POST(JSONmessageBuffer);
        if (runTimeVariables.serialOutRequestDetails == 1) {
          serialPrint(&Serial, F("\nSending POST Request: "));
          serialPrint(&Serial, runTimeVariables.i2cPostUrl.c_str());
        }
        
        if (lastTxResponse > 0) {
          if (runTimeVariables.serialOutRequestDetails == 1) {
            serialPrint(&Serial, F(" HTTP Response: "));
            serialPrint(&Serial, (String)lastTxResponse);
          }
          
          if (lastTxResponse == runTimeVariables.httpGoodResponse) {
            if (runTimeVariables.serialOutRequestDetails == 1) {
              String payload = http.getString();
              serialPrint(&Serial, F("\nResponse Body: \n"));
              serialPrint(&Serial, payload);
            }
          }
        } else {
          serialPrint(&Serial, F("\nHTTP Error: "));
          serialPrint(&Serial, http.errorToString(lastTxResponse).c_str());
        }
        http.end();
      }
    }
  }
}
```

* compiler output:
```
i2c_scan.ino: In function 'void txI2CData()':
i2c_scan:821:21: error: 'ArduinoJson::JsonObject' has no member named 'prettyPrintTo'
         JSONencoder.prettyPrintTo(JSONmessageBuffer, sizeof(JSONmessageBuffer));
```

Slyke, 2019-05-10:
After going through the source code it seems it has been removed and replaced with `serializeJsonPretty()`. It's still in the v6 documentation though: https://arduinojson.org/v6/api/jsonobject/createnestedarray/

bblanchon, 2019-05-10:
Hi @Slyke,

Yes, that was an error in the documentation.
Thank you very much for pointing that out.

Best Regards,
Benoit

closed 2019-05-10 by bblanchon