bblanchon/ArduinoJson #988: Empty document serialization in file

Jerkoco, 2019-05-10:
Hi,
I'm doing an Air Quality Station with a web interface. I'm using ESP01 (ESP8266) and Arduino Mega 2560
I'm using ArduinoJSON 6.9.1 on both cards.

The problem is:

I want to save my data to an SD card.
If the file of the day doesn't exist, I create it and write in the firsts measurements.
If the file of the day already exist, I open it, deserialize, add new measurements, and reserialize in.

When the 2nd step happen, I get a blank file.

here is my code:

`
DynamicJsonDocument savedoc(3000);

      Serial.println(freeRam());
      if (!SD.exists(idate + ".txt")) {
        // Si il n'existe pas encore de fichier à la date du jour, on en créé un
        myFile = SD.open(idate + ".txt", FILE_WRITE);
        //ajouter temperature et humidite
        savedoc["start"] = ihour;
        savedoc["qualiteair"].add(sensor.getValue());
        savedoc["particules"].add(concentration);
        serializeJson(savedoc, myFile);
        myFile.close();
        savedoc.clear();
      }  else {
        
        myFile = SD.open(idate + ".txt", FILE_WRITE);
        deserializeJson(savedoc, myFile);
        delay(500);
        myFile.close();
        delay(500);
        SD.remove(idate + ".txt");
        savedoc["qualiteair"].add(sensor.getValue());
        savedoc["particules"].add(concentration);
        SD.open(idate + ".txt", FILE_WRITE);
        serializeJson(savedoc, myFile);
        myFile.close();
        savedoc.clear();
        
      }
`
Any idea why ? Thanks a lot. 

P-S: I saw in https://github.com/bblanchon/ArduinoJson/issues/969
that you recommend using HTTP 1.0 to serve JSON, but didn't understand why.
My project use HTTP 1.1. Will this cause any problem with ArduinoJSON ?

Thanks.

bblanchon, 2019-05-13:
Hi @Jerkoco,

This looks wrong:

```c++
myFile = SD.open(idate + ".txt", FILE_WRITE);
deserializeJson(savedoc, myFile);
```

Please try with `FILE_READ`.

Best Regards,
Benoit

Jerkoco, 2019-05-13:
Hi, I changed this and I had:
SD.open instead of myFile = sd.open.

Thanks a lot Benoit.
Why do recommend not using http 1.1 with ArduinoJson ? Thanks.

bblanchon, 2019-05-14:
Hi @Jerkoco,

I recommend HTTP 1.0 because it avoids [Chunked Transfer Encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding).

Best Regards,
Benoit

Jerkoco, 2019-05-14:
Thanks a lot. Have a good day.

bblanchon, 2019-05-15:
You’re welcome. Thank you for using ArduinoJson.

closed 2019-05-15 by bblanchon