bblanchon/ArduinoJson #989: passing doc from function to function

mars000, 2019-05-12:
Hi
Im using V6.10.1 of this great library.
I'm trying to determine how to pass doc to function as a pointer so I save on stack/memory.
I'm kind of stuck as you can see below. Not sure how to do this gracefully. Any help would be appreciated :-)

1. In one function I have the following:
```cpp
DynamicJsonDocument doc(5000);

	// Parse JSON object
	DeserializationError error = deserializeJson(doc, client_json);

if (json_has_all_required_keys(&doc))
{ .... }
```

then I have:

```cpp
bool json_has_all_required_keys(DynamicJsonDocument *local_doc)
{
	bool keys_all_ok = true; // assume json file has all the expected json KEYS
	
        char *error;

	error = *local_doc["fileversion"];  // <--- PRODUCES COMPILER ERRORS
	if (error)
	{
		return (false);
	}

	if (!local_doc->containsKey("fileversion"))  // <-- THIS WORKS
	{
		return (false);
	}
```

ALSO
How would I reference a NESTED element using ```doc->containesKey(" ");```

bblanchon, 2019-05-13:
Hi @mars000,

*References to the rescue!*

Remember: a reference is a pointer with value syntax.
Here is what you're look for:

```c++
if (json_has_all_required_keys(doc)) // ...

bool json_has_all_required_keys(const JsonDocument& local_doc)
{
}
```

You could use `DynamicJsonDocument&`, but using `JsonDocument&` supports `StaticJsonDocument` too. Also, since you don't modify the document in the function, you can use `const`.

Lastly, the compiler error, is probably [invalid conversion from 'const char*' to 'char*' [-fpermissive]](https://arduinojson.org/v6/faq/error-invalid-conversion-from-const-char-to-char/), because you forgot `const` in the declaration of `error`.

Best Regards,
Benoit

mars000, 2019-05-14:
thank you - all working nicely now.

closed 2019-05-14 by mars000

bblanchon, 2019-05-14:
You're welcome @mars000.
Thanks again for using ArduinoJson.