bblanchon/ArduinoJson #990: ArduinoJson V6  -> JsonObject createNestedObject

mjoelmendes, 2019-05-13:
Hi, I'm using:
-Arduino 1.89
-ArduinoJson.h (version 6.10.1)
-Pacaca Generic ESP8285
-Upload Speed: "115200"

I'm trying to create a json with the following format:

{
   "aaa": {
      "bbb": "XXX",
      "ccc": "YYY"
   }
}

using the following code:

#include <ArduinoJson.h>

String bbb = "XXX";
String ccc = "YYY";

#define USE_SERIAL Serial
void setup () {
  // put your setup code here, to run once:

  USE_SERIAL.begin (57600);
  USE_SERIAL.print (F ("Booting on"));
  USE_SERIAL.println (ARDUINO_BOARD);

}

void loop () {
  // put your main code here, to run repeatedly:

  const size_t BUFFERSIZE = JSON_OBJECT_SIZE (1) + JSON_OBJECT_SIZE (2);
  DynamicJsonDocument jsonBuffer (BUFFERSIZE);

  JsonObject aaa = jsonBuffer.createNestedObject ("aaa");
  aaa ["bbb"]. set (bbb);
  aaa ["ccc"]. set (ccc);
/ *
  or
  aaa ["bbb"] = (bbb);
  aaa ["ccc"] = (ccc);
 */

  serializeJsonPretty (jsonBuffer, Serial);

  delay (5000);

}

the result that is given to me is the following:

{
  "aaa": {
    "bbb": "XXX"
  }
}

ccc field does not appreciate me and I can not see why someone can help me?

bblanchon, 2019-05-14:
Hi @mjoelmendes,

The `JsonDocument` is too small.
See:

* [Why is the output incomplete?](https://arduinojson.org/v6/faq/why-is-the-output-incomplete/)

Best Regards,
Benoit

mjoelmendes, 2019-05-14:
Hi @bblanchon, thanks.

closed 2019-05-14 by mjoelmendes