bblanchon/ArduinoJson #991: Floating point results in integer when writing 0.0 number

sdiaz-qkt, 2019-05-15:
I know there was a thread about this in [https://github.com/bblanchon/ArduinoJson/issues/22 ](url), but it does not solve my issue, as it is still happening even after last release (I'm using version 6.10.1).
I am simply trying to write values like 0.0, 2.0, etc inside my json. However, it keeps translating them to integers like 0, 2, etc. Since my application requires exclusively float values,  my application fails when the value I'm sending is an integer.

Any workaround for this ? 

Thank you.

bblanchon, 2019-05-18:
Hi @sdiaz-qkt,

I answered your question in the FAQ:

* [How to configure the serialization of floats?](https://arduinojson.org/v6/faq/how-to-configure-the-serialization-of-floats/)

Best Regards,
Benoit

sdiaz-qkt, 2019-05-20:
Hi, thanks for answering (and congrats for reaching 1000 issues btw!). There is no way that the serializer translate my floating 0.0 into "0.0" instead that "0"?

sdiaz-qkt, 2019-05-20:
Oh, nevermind. I'm an idiot and just finished reading the link you sent me. Is working fine now. Thank you!

closed 2019-05-20 by sdiaz-qkt

bblanchon, 2019-05-21:
You're welcome @sdiaz-qkt.
Thank you for using ArduinoJson.