bblanchon/ArduinoJson #993: Convert arrays to vectors

Daemach, 2019-05-16:
Can a struct contain a vector?  How do I instantiate that?

I have a nested array that could have 1-20 elements.  I would prefer to create a vector so it can be dynamic, rather than create a 2D array that could only have 1 element populated.  How would I populate this vector using arduinojson?

bblanchon, 2019-05-16:
Hi @Daemach,

ArduinoJson is not well integrated with STL containers, mainly because the STL is rarely available on Arduino. I hope I'll be able to improve that.

Anyway, you lost me at "2D array"; can you write some pseudo code, so I can clearly understand what you want?

Best Regards,
Benoit

Daemach, 2019-05-16:
I have a set of rules that I use to control devices.  The user can enter any number of rules.  This snippet has but 3:

	"rules": [
		[3.0, 9000, 10000, 254, 0.0, 0.0, 0.0, 0, 0],
		[3.0, 1500, 1250, 255, 0.0, 1.0, 0.0, 0, 0],
		[3.0, 1000, 1200, 255, 0.0, 0.0, 0.0, 0, 0]
	]

At the moment in my config object I am creating  the following as placeholders - this object will only hold 12 rules.  It would be nice to be able to use a vector here so I can add as many or as few rules as I need.
typedef struct {  // config struct
uint16_t aRules[12][7]      = {{0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}, {0, 0, 0, 0, 0, 0, 0}};
} configObject;  //end struct config

This isn't a big deal.  I can live with what I have.  It's just not that elegant, but neither am I so that's OK ;)

Daemach, 2019-05-16:
I am new to C btw - I don't even know how to add a vector to a struct.  I just know that it's more dynamic than an array and therefore I would like to figure it out.

bblanchon, 2019-05-25:
Assuming you can use `std::vector`, you could define a `rule` structure and use an `std::vector<rule>`.

But `std::vector` is only available in a few cores, and I think that's a good thing because using dynamic containers on a microcontroller is too risky. It's safer to allocate a fixed-size array for the largest of your use cases, than waiting for a `bad_alloc` exception to crash you program.

closed 2019-06-06 by bblanchon