bblanchon/ArduinoJson #994: Strings are killing me :/

Daemach, 2019-05-16:
I have a config struct with a lot of variables.  Everything is working except these 2 objects which I can't get to compile in any way.  I really don't care what the datatypes in the struct end up as, as long as I can use them as a string or char array.  Can you tell me what I need to do to get these json objects into the struct variables?

//Config struct

  typedef struct {  // config struct
    char accountKey[37];
    char APList[2][36];
  } configObject

//JSON snippet
{
"accountKey": "8F9457C9-E393-47F8-A57C-30E2BB70E026",
"APList": [
    ["MyAP1", "MyPW1"],
    ["MyAP2", "MyPW2"]
  ]
}

bblanchon, 2019-05-19:
Hi @Daemach,

This looks correct, can you share some actual code?

Best Regards,
Benoit

Daemach, 2019-05-19:
I got the account key figured out.  I had the wrong key name in the json call and must have passed a null to strlcpy which blew it up.  

  for (int i = 0; i < root["APList"].size(); i++) {
      for (int j = 0; j < root["APList"][i].size(); j++) {
        strlcpy(config.APList[i][j], root["APList"][i][j], sizeof(config.APList[i][j]));
      }
    }

Throws

configFunctions.h:133:36: error: invalid conversion from 'char' to 'char*' [-fpermissive]

         strlcpy(config.aAPList[i][j], root["aAPList"][i][j], sizeof(config.aAPList[i][j]));

Daemach, 2019-05-20:
Changing the state struct value to char* aAPList[2][36]; got past the compiler error, but it crashes the stack when it tries to do the strlcpy, again acting as if the value was null.

I tried the following but get a compile error:
```
      for (int i = 0; i < root["aAPList"].size(); i++) {
         for (int j = 0; j < root["aAPList"][i].size(); j++) {
            Serial.println(root["aAPList"][i][j]);
            if (root["aAPList"][i][j] != NULL) {
               strlcpy(config.aAPList[i][j], root["aAPList"][i][j], sizeof(config.aAPList[i][j]));
            }
         }
      }
```

call of overloaded 'println(ArduinoJson6101_000::ElementProxy<const ArduinoJson6101_000::ElementProxy<const ArduinoJson6101_000::MemberProxy<ArduinoJson6101_000::JsonDocument&, const char*>&>&>)' is ambiguous

bblanchon, 2019-05-25:
Hi @Daemach,

I answered your question in the FAQ:

* [Error: call of overloaded 'println(...)' is ambiguous](https://arduinojson.org/v6/faq/error-call-of-overloaded-println-is-ambiguous/)

Best Regards,
Benoit

Daemach, 2019-05-26:
That does solve the println problem.   Dumping those values using the fixed
println in the following code shows that they are NOT NULL.

strlcpy still crashes without the NULL check here, so I believe that null
check is returning TRUE, and root["APList"][i][j].as<const char*>() is also
returning NULL.  But how can that be, if the println directly before that
returns string values?

         for (int i = 0; i < root["APList"].size(); i++) {
            for (int j = 0; j < root["APList"][i].size(); j++) {
               Serial.println(root["APList"][i][j].as<String>());
              // if (root["APList"][i][j] != NULL) {
                  strlcpy(config.APList[i][j],
root["APList"][i][j].as<const char*>(), sizeof(config.APList[i][j]));
              // }
            }
         }

Daemach, 2019-06-03:
Any word on this?

bblanchon, 2019-06-05:
Hi @Daemach,

Please apologize for the late answer.
You can (and should) protect yourself against null strings by using [JsonVariant::operator|](https://arduinojson.org/v6/api/jsonvariant/or/):

```c++
strlcpy(destination, doc["key"] | "default", sizeof(destination));
```

Of course, you can replace `"default"` with any string you want.

Best Regards,
Benoit

Daemach, 2019-06-05:
Yes.  

My point was that the string is NOT null because the println printed it.  Something is broken.

My json file contains values.  I still cannot copy them into my state struct.  Please tell me what I need to do to make this work.

Daemach, 2019-06-08:
Strings are BROKEN in arduinoJSON 6.  The following code never runs strlcpy but it DOES print the strings out using println just before the if statement.  Therefore those keys are NOT null and your null check is FAILING.  And when I remove the null check, strlcpy fails!

`
for (int i = 0; i < root["APList"].size(); i++) {
    for (int j = 0; j < root["APList"][i].size(); j++) {
        Serial.println(root["APList"][i][j].as<String>());
        if (root["APList"][i][j] != NULL) {
            strlcpy(config.APList[i][j], root["APList"][i][j].as<const char*>(), sizeof(config.APList[i][j]));
        }
    }
}
`

bblanchon, 2019-06-10:
Hi @bblanchon,

`as<String>()` and `as<const char*>()` are quite different: the first calls `serializeJson()` whereas the second returns a pointer to the internal buffer. So it's possible that the first returns something and the second returns `NULL`, for example, if the variant contains an integer.

[`NULL` is not what you think is](https://cpp4arduino.com/2018/10/26/why-cpp-programmers-dont-use-null.html). What you wrote is `root["APList"][i][j] != 0`, which is not what you want. You must use `nullptr` or [JsonVariant::isNull()](https://arduinojson.org/v6/api/jsonvariant/isnull/) instead.

Best Regards,
Benoit

Daemach, 2019-06-10:
I have a config struct with a lot of variables. Everything is working except these 2 objects which I can't get to compile in any way. I really don't care what the datatypes in the struct end up as, as long as I can use them as a string or char array. Can you tell me what I need to do to get these json objects into the struct variables?
`
//Config struct

typedef struct { // config struct
char accountKey[37];
char APList[2][36];
} configObject

//JSON snippet
{
"accountKey": "8F9457C9-E393-47F8-A57C-30E2BB70E026",
"APList": [
["MyAP1", "MyPW1"],
["MyAP2", "MyPW2"]
]
}
`

bblanchon, 2019-06-15:
Hi @Daemach,

The `APList` array lacks one dimension.
It should be:

```c++
char APList[2][2][36];
```

See the complete demo here:
https://wandbox.org/permlink/on7bSlcdyzvUFv5K

BTW, the last chapter of [Mastering ArduinoJson](https://arduinojson.org/book/) shows how to serialize complex JSON configurations objects.

Best Regards,
Benoit

closed 2019-06-26 by bblanchon