bblanchon/ArduinoJson #995: ESP32 ... Randomly, the loop runs more times than expected.

clydde, 2019-05-16:
Hi, at ESP32 + ArduinoJson 6.10.1 + deserialize in chunks + SPIFFS + StreamUtils 1.2.0- problem:

Randomly, the loop runs more times than expected.

clydde, 2019-05-17:
It turns out that I have a "if" within the "do" of the "parse in chunks" that compares a key that when it is matched, it will activate a relay, and it will exit the loop with a "break".

It worked well in version 6.10, and most times the relay is operated as expected. but (randomly) the relay is activated up to 2/3 times.

As if sometimes the loop ignores the break or the json's output.

clydde, 2019-05-17:
In addition, reset problems like this have also appeared:

Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400da8b2  PS      : 0x00060830  A0      : 0x800da8fe  A1      : 0x3ffb1e70  
A2      : 0x3f401830  A3      : 0x00000000  A4      : 0x3ffb1e70  A5      : 0x00000010  
A6      : 0x3ffb1eac  A7      : 0x3ffb1e30  A8      : 0x800da8b2  A9      : 0x3ffb1e50  
A10     : 0x00009816  A11     : 0x3ffb1e70  A12     : 0x0000000c  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000003  
EXCVADDR: 0x3f40183c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400da8b2:0x3ffb1e70 0x400da8fb:0x3ffb1e90 0x400da953:0x3ffb1ec0 0x400d2b66:0x3ffb1ee0 0x400da209:0x3ffb1f30 0x400da264:0x3ffb1f70 0x400d1fd8:0x3ffb1f90 0x400db9ad:0x3ffb1fb0 0x4008877d:0x3ffb1fd0

Rebooting...

bblanchon, 2019-05-17:
Hi @clydde,

Please share a **small** code snippet and run the [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder).

Best Regards,
Benoit

clydde, 2019-05-19:
Thank you, 

**This is the code:**
`void recerca(String numCard2) {
  
// To deactive the RFID Hardware 
 detachInterrupt(digitalPinToInterrupt(PIN_D0));
  detachInterrupt(digitalPinToInterrupt(PIN_D1));

  numCard2.toUpperCase();

  File myFile = SPIFFS.open("/users.txt", "r");
  ReadBufferingStream bufferedFile(myFile, 126);

  if (!bufferedFile.find("\"usr\":")) {
    Serial.println(F("Array \"usr\" no s'ha trobat"));
  }
  DynamicJsonDocument doc(126);
  
  bufferedFile.find("[");

  do {
    auto err = deserializeJson(doc, bufferedFile);
    if (err) {
      Serial.print(F("deserializeJson() failed with code "));
      Serial.println(err.c_str());
     //To reactive the RFID Hardware 
      attachInterrupt(digitalPinToInterrupt(PIN_D0), pinStateChanged, CHANGE);
      attachInterrupt(digitalPinToInterrupt(PIN_D1), pinStateChanged, CHANGE);
      return;
    }

    const char* latar = doc["tar"].as<char*>();

    String latardec = decripta(latar);

    if (latardec == numCard2) {
      digitalWrite(32, LOW);
      wait(1000);
      digitalWrite(32, HIGH);
      guardaHora(latar);
      break;
    }

  } while (bufferedFile.findUntil(",", "]"));
  
  myFile.close();
//To reactive the RFID Hardware 
  attachInterrupt(digitalPinToInterrupt(PIN_D0), pinStateChanged, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_D1), pinStateChanged, CHANGE);

}`

### This is the debug info:

**Serial Monitor say:**
`Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400edfc6  PS      : 0x00060e30  A0      : 0x800eda84  A1      : 0x3ffb1ad0  
A2      : 0x00000000  A3      : 0x3ffb1cc4  A4      : 0x00000000  A5      : 0x400d1c70  
A6      : 0x3ffb1e60  A7      : 0x00000008  A8      : 0x3ffb1de0  A9      : 0x3ffb1ac0  
A10     : 0x3ffb1ae0  A11     : 0x3ff9c510  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000019  EXCCAUSE: 0x00000003  
EXCVADDR: 0x400d1c70  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400edfc6:0x3ffb1ad0 0x400eda81:0x3ffb1de0 0x400d1ffb:0x3ffb1ea0 0x400d2aee:0x3ffb1ee0 0x400da211:0x3ffb1f30 0x400da26c:0x3ffb1f70 0x400d1fd8:0x3ffb1f90 0x400db9b5:0x3ffb1fb0 0x4008877d:0x3ffb1fd0`

**And Exception Decoder Say**:
`PC: 0x400edfc6: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 925
EXCVADDR: 0x400d1c70: pinStateChanged() at C:\Users\webmaster\Documents\Arduino\CSMS\WiegandAnotherProves10/WiegandAnotherProves10.ino line 104

Decoding stack results
0x400edfc6: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 925
0x400eda81: sprintf at ../../../.././newlib/libc/stdio/sprintf.c line 644
0x400d1ffb: decripta(char const*) at C:\Users\webmaster\Documents\Arduino\CSMS\WiegandAnotherProves10/WiegandAnotherProves10.ino line 194
0x400d2aee: recerca(String) at C:\Users\webmaster\Documents\Arduino\CSMS\WiegandAnotherProves10/WiegandAnotherProves10.ino line 158
0x400da211: Wiegand::flushData() at C:\Users\webmaster\Documents\Arduino\libraries\Yet_Another_Arduino_Wiegand_Library\src\Wiegand.cpp line 196
0x400da26c: Wiegand::flush() at C:\Users\webmaster\Documents\Arduino\libraries\Yet_Another_Arduino_Wiegand_Library\src\Wiegand.cpp line 251
0x400d1fd8: loop() at C:\Users\webmaster\Documents\Arduino\CSMS\WiegandAnotherProves10/WiegandAnotherProves10.ino line 98
0x400db9b5: loopTask(void*) at C:\Users\webmaster\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\main.cpp line 19
0x4008877d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143`

bblanchon, 2019-05-26:
Hi @clydde,

The crash occurs in `decripta()` when you call `sprintf()`... maybe `latar` is null.
You can use [ArduinoTrace](https://github.com/bblanchon/ArduinoTrace) to dump the values of variables.

Best Regards,
Benoit

clydde, 2019-05-27:
Thank you, at this time I am doing the project's electronical tasks (which are complicated as well), soon I will put it back and tell you how it has gone. THANK YOU SO MUCH!

closed 2019-06-06 by bblanchon