bblanchon/ArduinoJson #996: Getting looping JSON

Slyke, 2019-05-17:
Not sure why this is happening. I have the following code running on an ESP8266:
```
void ICACHE_FLASH_ATTR httpHandleNotFound() {
  StaticJsonDocument<512> JSONbuffer;
  char JSONmessageBuffer[512];

  JsonObject JSONencoder = JSONbuffer.to<JsonObject>();
  JSONencoder["payload"] = JSONbuffer.to<JsonObject>();

  serializeJson(JSONencoder, JSONmessageBuffer);
  Serial.println(JSONmessageBuffer);
  webServer.send(404, "application/json", JSONmessageBuffer);
}
```

I would expect the output to be:
`{payload:{}}`, but instead the ESP8266 crashes, and the serial prints:
```
{"payload":{"payload":{"payload":{"payload":{"payload":{"payload":{"payload":{"payload":{"payload":{"payload":{"payload":{"payload":{"paylo
```
Before the stack trace starts.

If I change the code to:
```
void ICACHE_FLASH_ATTR httpHandleNotFound() {
  StaticJsonDocument<512> JSONbuffer;
  char JSONmessageBuffer[512];
  
  JsonObject JSONencoder = JSONbuffer.to<JsonObject>();
  JSONencoder["payload"] = JSONbuffer.to<JsonObject>();
  JSONencoder["payload"]["meta"] = JSONbuffer.to<JsonObject>();
  JSONencoder["payload"]["meta"]["uptime"] = millis();

  serializeJson(JSONencoder, JSONmessageBuffer);
  Serial.println(JSONmessageBuffer);
  webServer.send(404, "application/json", JSONmessageBuffer);
}
```

I get this before the stacktrace starts:
`{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{"payload":{"meta":{}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}`

I have similar code running on another project, and it's working fine (but the ESP8266 is sending a POST on the working version, in this project the ESP8266 is replying to a GET request).

Slyke, 2019-05-17:
Seems like the `JSONbuffer.to<JsonObject>()` is no longer needed for nested objects. I've changed my code to this:
```
void ICACHE_FLASH_ATTR httpHandleNotFound() {
  StaticJsonDocument<512> JSONbuffer;
  char JSONmessageBuffer[512];
  
  JsonObject JSONencoder = JSONbuffer.to<JsonObject>();
  JSONencoder[F("payload")];
  JSONencoder[F("payload")][F("meta")];
  JSONencoder[F("payload")][F("status")];

  JSONencoder[F("payload")][F("meta")][F("uptime")] = millis();

  JSONencoder[F("payload")][F("status")][F("code")] = "404";
  JSONencoder[F("payload")][F("status")][F("codeMessage")] = "Not Found";

  serializeJson(JSONencoder, JSONmessageBuffer);
  
  webServer.send(404, "application/json", JSONmessageBuffer);
}
```

bblanchon, 2019-05-19:
Hi @Slyke,

Indeed, the following line created a recursive object:

```c++
JSONencoder["payload"] = JSONbuffer.to<JsonObject>();
```

I can see the misunderstanding: you thought that [`JsonDocument::to<JsonObject>()`](https://arduinojson.org/v6/api/jsondocument/to/) was the equivalent of [`JsonBuffer::createObject()`](https://arduinojson.org/v5/api/jsonbuffer/createobject/). These functions look similar because the both return a new object, but [`JsonDocument::to<JsonObject>()`](https://arduinojson.org/v6/api/jsondocument/to/) clears the document because a [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) can only have one root.

As you saw, version 6 is able to create the objects as need.
You could reduce this function to simply:

```c++
void ICACHE_FLASH_ATTR httpHandleNotFound() {
  StaticJsonDocument<512> doc;  
  doc[F("payload")][F("meta")][F("uptime")] = millis();
  doc[F("payload")][F("status")][F("code")] = "404";
  doc[F("payload")][F("status")][F("codeMessage")] = "Not Found";

  char JSONmessageBuffer[512];
  serializeJson(JSONencoder, JSONmessageBuffer);
  
  webServer.send(404, "application/json", JSONmessageBuffer);

```

Best Regards,
Benoit

closed 2019-05-26 by bblanchon