bblanchon/ArduinoJson #998: jSonDocument property: sometimes null, sometimes not!

vaz83, 2019-05-19:
Hi,
i have the following code:
`void read()
{
 if (SPIFFS.begin()) 
  {
    if(SPIFFS.exists("/dev.json"))
    {
      File infoFile = SPIFFS.open("/dev.json", "r");
      if(infoFile)
      {
        size_t size = infoFile.size();
        std::unique_ptr<char[]> buf(new char[size]);
        infoFile.readBytes(buf.get(), size);

        DynamicJsonDocument jDoc(1500);
        DeserializationError error = deserializeJson(jDoc,buf.get());
        if(!error)
        {
          if(jDoc["GUID"]!=NULL)
          {
              String tempstr = jDoc["GUID"];
              if(tempstr.startsWith("SECURED."))
              {
                tempstr.replace("SECURED.","");
                jDoc["GUID"] = tempstr;
                tempstr.toCharArray(GUID,80);
                XORcrypt(GUID);
              }
              else
                strcpy(GUID, jDoc["GUID"]);
          }
        }
      }
    }
  }`

What i am facing now is that the condition if(jDoc["GUID"]!=NULL) is returning false, and when i pass jDoc["GUID"] to a string variable it has value! the property exists and has data. Why is it returning null? in another part of the code when i open other json files this condition works just fine, please help

bblanchon, 2019-05-19:
Hi @vaz83,

The problem is that [NULL is not a pointer](https://cpp4arduino.com/2018/10/26/why-cpp-programmers-dont-use-null.html), it is the integer `0`, so ArduinoJson applies the conversion to integer. Since the value in `doc["GUID"]` is not an integer, the conversion returns 0, so the `if` clause runs.

I'll fix the comparison operator so that it returns `false` when the value has the wrong type. I'll also to add support for nullptr_t, so you can write `if (doc["GUID"] != nullptr)` instead.

Best Regards,
Benoit

bblanchon, 2019-05-19:
As a temporary workaround, please use [JsonObject::containsKey()](https://arduinojson.org/v6/api/jsonobject/containskey/)

vaz83, 2019-05-19:
> As a temporary workaround, please use [JsonObject::containsKey()](https://arduinojson.org/v6/api/jsonobject/containskey/)

Will do. I also found out other problem, i will make a new thread. Thanks!

UltimatPronin, 2019-05-22:
I have similar problem (on V6):
1)in my sketch I use an DynamicJsonDocument _item(256);
2)periodically I change data stored in this object and send json to my service:
      _item["last_time"] = timeClient.getFormattedTime(); //string data   
      _item["rnd"] = random(2); //int data
and after about 100 operations of changing _item["last_time"] it continiously return null and I can not assign any value to it.
3)For now a workaround for me is pass [long type] variable [string] place to _item["last_time"]. With it all is ok.

commit 2ed77d2c (2019-05-26): Added support for `nullptr` (closes #998)
closed 2019-05-26 by bblanchon