bblanchon/ArduinoJson #1002: Failed to pass String (variable) to JSONObject

sarkrui, 2019-05-21:
I could pass **float**, **int** variables into a JSONObject, however, 
`objectData["time"] = formattedDate;`
I was able to complie but 
`Serial.println(logData);` outputs, this isn't an issue with **ArduinoJSONv5**.

> {"dBValue":56,"buttonState":2,"timeStamp":null}

### With ArduinoJSON v6
```
  //generated by arduinojson.org/v6/assistant/
  const size_t capacity = JSON_OBJECT_SIZE(3);
  DynamicJsonDocument objectData(capacity);

  objectData["dBValue"] = dBValue;
  objectData["buttonState"] = buttonState;
  objectData["timeStamp"] = formattedDate;

  String logData;
  serializeJson(objectData, logData);
  Serial.println(logData);
  //Firebase.pushJSON(firebaseData, data_log + device_id, logData);
```

UltimatPronin, 2019-05-22:
I have similar problem (on V6):
1)in my sketch I use an DynamicJsonDocument _item(256);
2)periodically I change data stored in this object and send json to my service:
      _item["last_time"] = timeClient.getFormattedTime(); //string data   
      _item["rnd"] = random(2); //int data
and after about 100 operations of changing _item["last_time"] it continiously return null and I can not assign any value to it.
3)For now a workaround for me is pass [long type] variable [string] place to _item["last_time"]. With it all is ok.

bblanchon, 2019-05-26:
Hi @sarkrui,

ArduinoJson copies `String`s in the `JsonDocument`, so you need to increase the capacity.
You didn't have this problem with ArduinoJson v5 because `DynamicJsonBuffer` grew automatically when full.

Best Regards,
Benoit

closed 2019-06-06 by bblanchon