bblanchon/ArduinoJson #1003: v6 imcompatible with UniversalTelegramBot

Roman1820, 2019-05-22:
Hi,
Adding just 1 line:
#include <UniversalTelegramBot.h>
causes the compilation error:
"...\Arduino\libraries\UniversalTelegramBot\src\UniversalTelegramBot.cpp:705:23: error: 'keyboardBuffer' was not declared in this scope
DynamicJsonBuffer keyboardBuffer;  ".

Wemos D1 R1
ArduinoJson 6.10.1
Arduino IDE 1.8.9
But there is no such an error in version 5.13.4 of UniversalTelegramBot. However I need to use ArduinoJson 6.10.1.

bblanchon, 2019-05-26:
Hi @Roman1820 

Please see the [migration guide](https://arduinojson.org/v6/doc/upgrade/)

Best Regards,
Benoit

Roman1820, 2019-05-26:
Hi Benoit,
I know that page. I use ArduinoJson 6 instructions like "DynamicJsonDocument doc(2048);" etc. and everything works fine until I add:
#include <UniversalTelegramBot.h>
That causes the above mentioned error.
Best regards,
Roman

bblanchon, 2019-05-26:
Hi Roman,

Three solutions:

1. Use ArduinoJson 5 in your program
2. Update UniversalTelegramBot for ArduinoJson 6
3. Add [ArduinoJson-v5.13.5.h](https://github.com/bblanchon/ArduinoJson/releases/download/v5.13.5/ArduinoJson-v5.13.5.h) in `Arduino\libraries\UniversalTelegramBot\src` 

Using solution 3 will create a bigger program because you'll have two versions of the library.

Best Regards,
Benoit

Roman1820, 2019-05-26:
Hi Benoit,
Thank you for your answer!
ArduinoJson 5 wokrs with UniversalTelegramBot 1.1.0 and  UniversalTelegramBot has no upgrades yet.
I can't understand why  UniversalTelegramBot is incompatible with ArduinoJson 6?
Best regards,
Roman

bblanchon, 2019-06-06:
Hi @Roman1820,

ArduinoJson 6 became stable in [March 2019](https://arduinojson.org/2019/03/05/version-6-9-0/), UniversalTelegramBot simply didn't have the time to upgrade.

You should open an issue or a Pull Request there; I'm sure many users are waiting for it.
The upgrade progress is pretty simple; see https://arduinojson.org/v6/doc/upgrade/.

Best Regards,
Benoit

closed 2019-06-06 by bblanchon