bblanchon/ArduinoJson #1004: When the msgpack code equals 10, it will change lines.

kingsjl, 2019-05-23:
When the msgpack code equals 10, it will change lines.

`
#include<ArduinoJson.h>
void setup() {
  Serial.begin(115200);
}

void loop() {
  const size_t capacity = JSON_ARRAY_SIZE(1) + JSON_OBJECT_SIZE(5);
  DynamicJsonDocument doc(capacity);
  doc["decives"] =10;
  doc["mode"] = 127;
  doc["id"] = 1;
  JsonArray data = doc.createNestedArray("data");
  data.add(6);
  doc["uuid"] = "111111";
  serializeMsgPack(doc, Serial);
  Serial.print('\n');
 }  
`
Arduino IDE serial output :
`
⸮⸮decives
⸮mode⸮id⸮data⸮⸮uuid⸮111111
`
Version:ArduinoJson 6.10.1

bblanchon, 2019-05-26:
Hi @kingsjl,

In the [ASCII standard](https://en.wikipedia.org/wiki/ASCII), `10` means "line feed".
What you see is perfectly normal.

Best Regards,
Benoit

BrightLin, 2019-05-27:
@bblanchon ，but “10” in the data is number, a integer number, not ASCII data.   
I add number "1,2,3,4 ...9...11,12,..." is ok， but not "10".  I can't use "10" number in my protocol. Just like the follow struct: 
` { age: 10, name: "Tom", sex: "m"}`

bblanchon, 2019-06-11:
@BrightLin, MessagePack stores numbers in binary format, but the Serial Monitor interprets each byte as an ASCII character.
Again, what you see is normal.
If you want to see `10` in the output, you must use a string.

closed 2019-06-11 by bblanchon