bblanchon/ArduinoJson #1005: as<bool>() not working

Daemach, 2019-05-23:
I'm trying to parse an array of single char strings (because my server is a pita) into a boolean array.  The relevant portion looks like this:

"aPortDir":["0","0","1","1","1","1","1","1"]

This code turns them all to 0's

         for (int j = 0; j < root["aPortDir"].size(); j++) {
            config.aPortDir[j] = root["aPortDir"][j].as<bool>();
         }

config struct defines that array as 

boolean aPortDir[8]         = {false, false, false, false, false, false, false, false};

sparkplug23, 2019-05-24:
Perhaps related so appending my question here:

I changed from v5 to v6, and previously I had been testing for keys using>
if(root["name"].as<const char*>()){}

Rather than first setting this directly to a const char* type. It works, for ALL datatypes. 

In v6, neither as<T> or is<T> seems to be working for me. Wondering why (what has changed?).

bblanchon, 2019-05-26:
@Daemach, ArduinoJson supports implicit conversion from string to bool, but only if the string contains `"true"`. Admittedly, it could also accept any non-null integer as `true`, but I need to check if this doesn't conflict with another rule and if it doesn't increase the size too much.

You can fix your program by calling `as<int>()` instead.

@sparkplug23, v6 and v5 are very different on that matter: v5 kept strings in memory and parsed the values lazily, whereas v6 parses values eagerly and discards input strings. That's why v6 returns a null here.

You can fix your program by calling [JsonObject::containsKey()](https://arduinojson.org/v6/api/jsonobject/containskey/) instead.

Best Regards,
Benoit

bblanchon, 2019-08-12:
I'll make `as<bool>()` return `true` for all values

sparkplug23, 2019-08-12:
@bblanchon Thanks for getting back to me in May!! I only just seen it because of your most recent comment. For the most part I have resolved any issues like this in my code. On that point, ArduinoJson lib is awesome! Amazing work!

I do wonder if there is anyway to neatly/correctly check contains key for nested jsons. I have quite a few, and I am currently following the documentation by reading as const char then testing.
I wonder if an equivalent containskey function could be used for multiple nexted keys? Perhaps with va_lists. I might actually create my own intermediate function to do it, but would be great if it was supported with arduinojson. e.g obj.containsKey("level1","level2", ...)

bblanchon, 2019-08-12:
The canonical way is `if (!obj["level1"]["level2"].isNull())`, but you can avoid the double lookup with an additional variable:

```c++
JsonVariant value = obj["level1"]["level2"];
if (!value.isNull()) {
   // ...
}
```

sparkplug23, 2019-08-12:
Great! Thanks for getting back to me. Although it requires multiple lookups I do prefer the visually cleaner canonical method.

I'll check myself, but don't suppose something like this work work:

`JsonVariant value = null; // or NULL? creating one variable for multiple if checks

if((value = obj["level1"]["level2"]) != null){ // set it, then check it
   //...
}
if((value = obj["level1"]["level2"]) != null){ // set it, then check it
   //...
}`

bblanchon, 2019-08-12:
`NULL` won't work, but `nullptr` will.

sparkplug23, 2019-08-22:
@bblanchon Hey sorry to bother you again. I was wondering if a "containskey" exists that is not case sensitive?

bblanchon, 2019-08-22:
No, there isn't a case-insensitive version of `containsKey()`.
However, you could write a case-insensitive `String` class, but it requires probably more work than you're willing to spend on that, and I cannot guarantee that it will continue working with future version.

@sparkplug23, please no more issue hijacking; next time, open a new issue.

closed 2019-08-29 by bblanchon

bblanchon, 2019-09-05:
Fix available in [ArduinoJson 6.12.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.12.0).

commit ed18e776 (2019-08-29): Changed as<bool>() to return true for any non-null value (closes #1005)