bblanchon/ArduinoJson #1006: ArduinoJson6 + MessagePack + UnrealEngine

mircobabini, 2019-05-24:
Hi Benoit, nice to meet you again after almost a year. I develop mixed reality products (like vr simulators) and I need a really fast, solid and flexible communication protocol between Unreal Engine and Arduino.

Already tried and hacked Firmata and many others, but it’s always the same: when fast, not flexible. When flexible, not solid.

That’s why I want to find the best way to do this. I mean doing a fast, solid and flexible communication protocol using ArduinoJson6 and MessagePack.

So, first question to match the “solid” requirement. How do you know that when you find something to read in serial port (Serial.available() == true) is not an incomplete string?

Should I read byte per byte and store into a buffer until a carriage return, then call the DeserializeJson?

Is there any serial message length limit to be aware of?

Thanks.

bblanchon, 2019-05-26:
Hi @mircobabini,

From my experience, waiting for `Serial.available() > 0`  is enough because ArduinoJson will read the message until the end and wait for the timeout if needed. If a timeout occurs, [deserializeMsgPack()](https://arduinojson.org/v6/api/msgpack/deserializemsgpack/) returns `IncompleteInput`.

I don't think you need to use a buffer if you're just reading from the serial port.

The Serial port doesn't put a limit on the size of the message, but be careful with the size of your `JsonDocument`.

Best Regards,
Benoit

mircobabini, 2019-05-26:
I see that `deserializeMsgPack` takes a Stream, so you are telling me to directly pass the Serial object. Seems good to avoid using a buffer, yes. Also, I looked at the source code (v6 of course) but I can't see where the timeout is managed or configurable. Looking at `deserialize.hpp` and `MsgPackDeserializer.hpp` with no luck.

Also, on the UE4 side, which is c++ as well, I'd love to use ArduinoJson too. Actually, I used a lot the UE4Duino plugin, free available on github. It uses this [Serial implementation](https://github.com/RVillani/UE4Duino/blob/master/UE4Duino/Source/UE4Duino/Private/Serial.h). Do you think is possibile to use the deserializeMsgPack() with it?

Thanks,
Mirco

bblanchon, 2019-05-29:
You should be able to use ArduinoJson there too.

`deserializeMsgPack()` supports the standard `std::istream`, but you can add support for virtually any stream by implementing a custom "reader" class; see [ArduinoStreamReader.hpp](https://github.com/bblanchon/ArduinoJson/blob/6.x/src/ArduinoJson/Deserialization/ArduinoStreamReader.hpp) for an example.

closed 2019-06-15 by bblanchon