bblanchon/ArduinoJson #1008: Erreur de compilation pour la carte LOLIN(WEMOS) D1 R2 & mini

bebenthefox, 2019-05-25:
Dear,

My first trialwithJSON and with arduinoJSON ... 
I have installed version 6.10.1, error at compilation time
To avoid a personal error from my coding, I Ihave simplified my sketch
void setup() {
#include <ArduinoJson.h>
}
void loop() {
}
I have reinstall version 5.13.5
Restart the arduino ide
Change the card from "LOLIN(WEMOS) D1 R2 & mini" to "Arduino/Genuino Uno"
The problem still the same ... could you help me ?

Thanks

Bernard

COMPILATION ERROR:
*****************************
In file included from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonBuffer.hpp:11:0,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:7,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:7,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:7,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\beben\OneDrive\Documents\ArduinoProjects\Learning\ESP8266\zzzzzzzz\zzzzzzzz.ino:2:

C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../Data/NonCopyable.hpp: In function 'void setup()':

C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../Data/NonCopyable.hpp:7:1: error: 'namespace' definition is not allowed here

 namespace ArduinoJson {

 ^

In file included from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../Data/JsonVariantContent.hpp:7:0,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariant.hpp:10,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonBuffer.hpp:12,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:7,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:7,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:7,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\beben\OneDrive\Documents\ArduinoProjects\Learning\ESP8266\zzzzzzzz\zzzzzzzz.ino:2:

C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../Data/JsonFloat.hpp:9:1: error: 'namespace' definition is not allowed here

 namespace ArduinoJson {

 ^

In file included from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../Data/JsonVariantContent.hpp:8:0,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariant.hpp:10,
[... 1184 lines cut ...]

In file included from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9:0,

                 from C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\beben\OneDrive\Documents\ArduinoProjects\Learning\ESP8266\zzzzzzzz\zzzzzzzz.ino:2:

C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:21:1: error: 'namespace' definition is not allowed here

 namespace ArduinoJson {

 ^

C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:161:27: error: expected '}' before end of line

 #pragma GCC diagnostic pop

                           ^

C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp: At global scope:

C:\Users\beben\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:161:27: error: expected declaration before end of line

exit status 1
Erreur de compilation pour la carte LOLIN(WEMOS) D1 R2 & mini

bblanchon, 2019-05-26:
Hi @bebenthefox,

You must put the `#include` statement at the top of the file, out of any scope.

```c++
#include <ArduinoJson.h>

void setup() {
}
void loop() {
}
```

Note that this is true for all libraries.

Best Regards,
Benoit

bebenthefox, 2019-05-26:
Thanks,

I'was tired ... or stupid !

Regards

Bernard

bblanchon, 2019-05-26:
No problem Bernard.
Thank you for using ArduinoJson.

closed 2019-05-26 by bblanchon