bblanchon/ArduinoJson #1009: Suggested example for you to add :)

TechMaxed, 2019-05-28:
This is another example for jsonserver. Hope you would like it :)
```
#include <ESP8266WebServer.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
const char* ssid     = "OdetoDeath";         //Wifi network
const char* password = "hotdogeggbeer33";  //Wifi password
#define temp 14
#define pot 12
ESP8266WebServer server; //server variable
void setup() {
  Serial.begin(115200);         // Start the Serial communication to send messages to the computer
  delay(10);
  Serial.println('\n');
  pinMode(temp,INPUT);
pinMode(pot,INPUT);
  WiFi.begin(ssid, password);             // Connect to the network
  Serial.print("Connecting to ");
  Serial.print(ssid); Serial.println(" ...");

  int i = 0;
  while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
    delay(1000);
    Serial.print(++i); Serial.print(' ');
  }

  Serial.println('\n');
  Serial.println("Connection established!");  
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());         // Send the IP address of the ESP8266 to the computer
server.on("/", []() {
String webPage;
StaticJsonDocument<256> doc;
JsonArray device = doc.createNestedArray("device");
JsonObject device_0 = device.createNestedObject();
device_0["temp"] = analogRead(temp);
device_0["pot"] = analogRead(pot);
serializeJson(doc, webPage);
  server.send(200, "text/html", webPage);
});
server.begin();
}
void loop() { 

}
```

bblanchon, 2019-05-28:
Hi @TechMaxed,

Thank you for this example, but I cannot add it to the "official" examples because it only works with ESP.

It's an excellent example that would be helpful for many beginners, it definitely needs to be public... maybe it's still to start your blog @TechMaxed ;-)

Best Regards,
Benoit

TechMaxed, 2019-05-29:
Thanks! I had a very hard time trying to achieve this since your example only includes ethernet. I posted this to save other people's time :)

closed 2019-06-11 by bblanchon