bblanchon/ArduinoJson #1011: Use extras sub-folder for ancillary content

rei-vilo, 2019-05-30:
Please place the sub-folders `fuzzing`, `scripts`, `third-party` and `test` on sub-folder `extras`, as per https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#extra-documentation.

> The content of the `extras` folder is totally ignored by the IDE; you are free to put anything inside such as supporting documentation, etc.

bblanchon, 2019-06-06:
Hi @rei-vilo,

Yes, that would certainly help alternative IDEs, like Sloeber Plugin, that had issues with those files.

It will certainly break some code, so I need to wait for a new major revision.

BTW, why do you need that?

Best Regards,
Benoit

rei-vilo, 2019-06-06:
I'm the developer of [embedXcode](https://embedxcode.weebly.com), a plug-in for Xcode to ease development for the most popular embedded computing boards.

I had to move the sub-folders previously listed into `extras` by hand to avoid any issue with embedXcode. 

+ General website https://embedxcode.weebly.com
+ User manual website https://embedxcode.com

closed 2019-09-03 by bblanchon

bblanchon, 2019-09-05:
Fix available in [ArduinoJson 6.12.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.12.0).

commit b47ac27a (2019-09-03): Moved ancillary files to `extras/` (fixes #1011)