bblanchon/ArduinoJson #1012: DynamicJsonDocument Question

czuvich, 2019-05-31:
I just have a quick question for DynamicJsonDocument.  I am happily using ArduinoJson v5, and I make heavy use of the DynamicJsonBuffer.  The reason I use this buffer is that I basically take advantage of the large heaps for the ESP32 and nRF52840.  I make calls to JSON backends (nRF52840 + ESP32 co-processor) that bring back a decently sized payload (5K or so); however, this payload could grow up to 10-20K in a production scenario.

Does the DynamicJsonDocument immediately allocate the buffer or does it grow UP to the maximum size?

bblanchon, 2019-06-04:
Hi @czuvich 

[DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/) has a fixed capacity: it allocates the buffer in the constructor and that's all; it doesn't grow up.

Best Regards,
Benoit

czuvich, 2019-06-04:
Thanks Benoit.  Could you help me understand the heap/stack allocation differences of using DynamicJsonDocument vs. StaticJsonDocument?  For example..

If I have 1 object with an array of 10 strings each 10 bytes long, how does the stack and heap allocation differ between the DynamicJsonDocument and StaticJsonDocument?  

I'm trying to figure out if I should move to Json 6 or stay on 5.  Great work on this library.

bblanchon, 2019-06-05:
Hi @czuvich,

[DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/) and [StaticJsonDocument](https://arduinojson.org/v6/api/staticjsondocument/) work the same way; the only difference is the location in the memory.
For small microcontrollers, like the UNO, use a StaticJsonDocument because it saves time and memory.
For larger ones, like the ESP8266, use a DynamicJsonDocument because the stack is limited to 4KB.
In both cases, use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) to compute the right capacity.

If you want to learn more, please read chapter 2 of [Mastering ArduinoJson](https://arduinojson.org/book/).

Best Regards,
Benoit

czuvich, 2019-06-05:
Hi Benoit,

I'm probably not being clear, but I'm trying really figure out the difference between DynamicJsonDocument and DynamicJsonBuffer.  Does your book discuss those differences?  I'd love to support your framework.

bblanchon, 2019-06-06:
Hi @czuvich,

For the differences between DynamicJsonDocument  and DynamicJsonBuffer, please check out the [Migration Guide](https://arduinojson.org/v6/doc/upgrade/):

> With ArduinoJson 5, it was very difficult to use a JsonObject or a JsonArray as a class member because you had to make sure that the JsonBuffer stayed in memory too. The trick was to add the JsonBuffer as a class member too, but it was more complicated than it should be.
>
> ArduinoJson 6 replaces the concept of JsonBuffer with the concept of JsonDocument. The JsonDocument owns the memory and contains the root of the object tree. You can see a JsonDocument as a combination of JsonBuffer and JsonVariant.
> [...]
> DynamicJsonDocument has a fixed capacity that you must specify to the constructor. Unlike the DynamicJsonBuffer, DynamicJsonDocument doesn’t automatically expand.

Thank you for considering supporting the project. However, [Mastering ArduinoJson](https://arduinojson.org/book/) doesn't compare DynamicJsonDocument and DynamicJsonBuffer: it focuses on version 6, and rarely mentions version 5.

Best Regards,
Benoit

czuvich, 2019-06-07:
Thanks Benoit.  I'd like to give some constructive feedback.

If I'm understanding this correctly, the DynamicJsonDocument isn't really dynamic (grows as needed), it's really a StaticJsonDocument that allocates memory to the heap.  Whereas, the DynamicJsonBuffer is dynamic since the buffer will allocate memory as needed.  I read the github thread that discussed why the dynamic feature was removed.

I'm currently using the ESP32 WROVER and nRF52840 chips to do various tasks, and these chips have quite a bit of memory in comparison to other Arduino variants.  I use DynamicJsonDocument knowing that memory will not get reclaimed; however, I don't have to think about running out of allocation buffer space.  All I need to worry about is to keep my serialization reasonably sized; however, a static allocation means I HAVE to worry about size (both upfront allocation AND serialization).

Since, in my case, the uC are running remotely and headless, the convenience of not having to think about memory issues (again within a reasonable limit) is extremely helpful.  Long story short, it would be nice to somehow maintain the dynamic growing of allocations for use cases such as my own; however, I understand that comes at a development/maintenance cost for yourself.  

Thanks for this excellent library!

bblanchon, 2019-06-08:
Thank you for the feedback @czuvich.

Yes, you're right. In v6, "dynamic" doesn't mean "elastic" anymore, but it still means "dynamic memory allocation." I hesitated to rename the two classes to `StackJsonDocument` and `HeapJsonDocument`, but this would still be inaccurate because the `StackJsonDocument` could be allocated in the heap. Also, I think that using similar names between v5 and v6, simplifies the upgrade process.

Engineering requires compromises, especially in embedded systems. To keep ArduinoJson simple, small, and performant on small and big microcontrollers, I decided to remove this feature.

To earn all of that, what did we lose? Not so much after all: the only difference is that you must specify the capacity. Is it so bad? I don't think so. On the contrary, I think it's excellent because it forces you to think about the worst case and to make your program more resilient.

After also, if you think you have enough memory on your microcontroller, why don't you allocate a gigantic `DynamicJsonDocument`?

closed 2019-06-09 by czuvich