bblanchon/ArduinoJson #1014: SUGGESTION: Continuous Fuzzing

yevgenypats, 2019-06-01:
Hi @bblanchon 

I saw that ArduinoJson is integrated with OSSFuzz though I'm not sure if it is active? if the integration is active and works then great.

If not, I'll be happy to write an integration to [Fuzzit](https://fuzzit.dev) (Our Continuous Fuzzing Platform) which is free for open-source projects.

Let me know if that's relevant for this project.

Cheers

bblanchon, 2019-06-06:
Hi @yevgenypats,

Yes, ArduinoJson is still running on OSS-Fuzz.

Fuzzit looks like a very good product, but the last thing I want is more code to maintain...
What does it require?

Best Regards,
Benoit

yevgenypats, 2019-06-06:
Hi @bblanchon ,

The integration is pretty simple it's just requires another step in Travis to send the compiled libFuzzer targets to Fuzzit.

I'll be happy to contribute and maintain this integration.

Let me know what you think.

Best,
Yevgeny

bblanchon, 2019-09-03:
Closing in favor of #1020.

closed 2019-09-03 by bblanchon