bblanchon/ArduinoJson #1016: Help with delay caused by parse attempt of empty JSON over SoftwareSerial

nova999, 2019-06-03:
Hello,
I have created a setup with an arduino nano and a nodemcu esp8266. I am using SoftwareSerial to have them communicate with each other. The esp hosts a web server, and upon user input it sends a JSON packet to the arduino.

My naive method of doing this was to check for a new json packet at the arduino side at each loop iteration -
`
void loop()
{
    DynamicJsonBuffer jsonBuffer(bufferSize);
    JsonObject& root = jsonBuffer.parseObject(s);
    if (root == JsonObject::invalid())
    {
      Serial.println(F("Error or no data received"));
     }
}
`
Where 's' is the SoftwareSerial Object.
On the other end the esp does `root.printTo(s);` where root is the json object.

This works fine except for the fact that  attempting to parse an empty JSON from the SoftwareSerial takes about 2 to 3 seconds. I've tested this in a test arduino sketch.

I have tried to do the following to solve this -

- use arduino's interrupt pins, the json is parsed and stored in a String object
- tried to use s.available() > 0 to detect that data has arrived
- tried to manually implement a pseudo interrupt (the arduino checks whether the esp sent a LOW signal)

But I was unable to make the above methods work properly.

I am using ArduinoJson version 5.13.4 as version 6 was still in beta when I wrote the sketch. The boards I'm using are not genuine boards, if that helps.

bblanchon, 2019-06-04:
Hi @nova999,

The delay is caused by a timeout in `SoftwareSerial`, you can configure it with [Stream::setTimeou()](https://www.arduino.cc/en/Reference/StreamSetTimeout).

I recommend that you check `s.available() > 0`.
I use this technique in the last case-study of [Mastering ArduinoJson](https://arduinojson.org/book/), except that I used `HardwareSerial` instead of `SoftwareSerial`.

```c++
  while (!Serial.available())
    delay(50);

  DynamicJsonBuffer jb;
  JsonVariant root = jb.parse(Serial);
```

Best Regards,
Benoit

nova999, 2019-06-07:
Thank you @bblanchon  for explaining that to me.

However, I did not mention this before so it's my fault, but the loop function is supposed to do other stuff after checking for a json object in serial, so I cannot use a while loop.

I previously tried to do something like -
``` C
String data;
...
...
void loop()
{
	if(s.available() > 0)
	{
		Serial.println("");
		DynamicJsonBuffer jsonBuffer(bufferSize);
		JsonObject& root = jsonBuffer.parseObject(s);
		if (root != JsonObject::invalid())
		{
			root.printTo(data);
		}
		else
		{
			Serial.println("Error parsing JSON data");
		}
	}
	...
	...
	//millis delay of 1 sec start
	if(data!=null)
	{
		DynamicJsonBuffer jsonBuffer(bufferSize);
	    	JsonObject& root = jsonBuffer.parseObject(data);
	    	//extract settings from json
	}
	//millis delay end
}
```
but I could not get it to work - the parse never succeeded even if I printed the value of s.available() and it said 63 which I think is the number of characters(?). If true then it matches with the length of the JSON.

I will try it again and let you know. If possible please let me know if the code logic itself is correct because then I can know that the problem may be elsewhere.

Thank you for you help.

bblanchon, 2019-06-15:
Hi @nova999,

This program looks OK, except that I would use `root.success()` instead of `root != JsonObject::invalid()`.
Maybe you should try with v6, because it would tell you why the deserialization failed.

Best Regards,
Benoit

nova999, 2019-06-18:
Hello, I actually got it working some how. Thank you for your help.

closed 2019-06-18 by nova999

bblanchon, 2019-06-18:
You're welcome @nova999. 
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project ;-)