bblanchon/ArduinoJson #1017: Parser discards duplicate keys, javascript comments, trailing characters, infers types in values of key value pairs

celdridge84, 2019-06-03:
target platform:  MIPS
compiler:  GCC 4.8.3

Hi,

Myself and my team are using ArduinoJson 5 as part of our development on a new feature.  I am currently in the process of migrating our project from ArduinoJson 5 to 6.  We have noticed some undesirable behaviors in the parsing mechanism in ArduinoJson 5 and wanted to see if you could offer a solution.  

#1:  trailing characters are discarded rather than recognized as an error (may be resolved by ArduinoJson 6)
-Request  = {'req_ver':'1.0', 'id':1, 'method':'my_method'}}
-Current Response  = {'req_ver':'1.0', 'id':1, 'result':'it worked!'}
-Desired Response  = ArduinoJson parse error due to extra '}'

#2:  Duplicate keys in a JSON object are not regarded as a parse error.  Second duplicate key replaces first.
-Request 1 = {'req_ver':'1.0', 'id':1, 'method':'try to confuse parser', 'method':'my_method'}
-Current Response 1  = {'req_ver':'1.0', 'id':1, 'result':'it worked!'}
-Desired Response  = ArduinoJson parse error due to duplicate key

-Request 2= {'req_ver':'1.0', 'id':1, 'method':'my_method', 'method':'try to confuse parser'}}
-Current Response 2  = {'req_ver':'1.0', 'id':1, 'error':'try to confuse parser not found!'}
-Desired Response  = ArduinoJson parse error due to duplicate key

#3:  Comments deleted from request rather than being a parse error.
-Request  = {'req_ver':'1.0', 'id':1,/*Restriction - Comment can't be used*/ 'method':'my_method'}
-Current Response  = {'req_ver':'1.0', 'id':1, 'result':'it worked!'}
-Desired Response  = ArduinoJson parse error due to comment but comment is simply removed from parsed JSON object.
https://stackoverflow.com/questions/244777/can-comments-be-used-in-json

#4:  The types in key value pairs are inferred if they are not correct
-Request  = {'req_ver':'1.0', 'id':1,'method':my_method} //notice that there are no quotes around my method.  ArduinoJson will infer my_method to be a JSON String anyways
-Current Response  = {'req_ver':'1.0', 'id':1, 'result':'it worked!'}
-Desired Response  = ArduinoJson parse error due to missing quotes

I can provide more concrete examples if needed.  Any help is greatly appreciated!

Thanks,
Chris

bblanchon, 2019-06-04:
Hi Chris,

ArduinoJson is not a strict parser; on the contrary, it tries to be as forgiving as possible.
This is by design and I don't plan to change that.

v5 and v6 are very similar, but you should see a difference in #1 and #2.

Best Regards,
Benoit

celdridge84, 2019-06-05:
Hi Benoit,

Thank you very much for your response.  I think we can work around the parsing behavior.

I do have one additional question for you related to ArduinoJson5.  I am trying to take advantage of the emplace parsing mechanism that JsonBuffer offers when the parse method a char*, but it doesn't seem to be working correctly.  I have a 64KB char buffer and 32KB StaticJsonBuffer but during parsing, it appears that only the StaticJsonBuffer is modified.  I see no escape sequences added to the 64KB char buffer if I look at it in memory after parsing is complete.

Is there something that I'm missing?

Thanks,
Chris

bblanchon, 2019-06-06:
Hi @celdridge84,

Make you that the right overload of [JsonBuffe::parseObject()](https://arduinojson.org/v5/api/jsonbuffer/parseobject/) is selected.
The overload that takes a `const char*` duplicates the input whereas the one taking a `char*` uses the zero-copy mode.

[JsonParserExample.ino](https://arduinojson.org/v5/example/parser/) uses the zero-copy mode.

Best Regards,
Benoit

closed 2019-06-15 by bblanchon