bblanchon/ArduinoJson #1018: Getting exceptions on an ESP8266. Not sure why.

Slyke, 2019-06-05:
Hello,

I'm writing an API that is running on an ESP8266 D1 Pro Mini. It's point is to be able to program an I2C EEPROM from REST calls that contain JSON. I have a serial version of the parser that uses similar logic, and the same functions running inside the project that is working flawlessly.

The full code can be found here: https://github.com/Slyke/EEPROM-Programmer/blob/master/src/eeprom_prog/http_handle.ino

The code is quite lengthy, but I've narrowed it down to my function that handles my HTTP request. Running this code crashes, I need to have `Block 1` and `Block 2` commented out (and obviously hard code in the commands I want to run, or pass them in from the serial port) and it works fine.

The exceptions I get are Exception 2 and Exception 9. Sometimes it just freezes and reboots with no exception. I can paste the full stack trace if anyone is interested. I've been on this problem for a few days trying to track down in the code where it is. It's possible I have a memory leak, not using ArduinoJSON correctly, or writing over some important memory and not realising it, but it seems to only crash when using ArduinoJSON. I do know that it is working correctly when I don't use ArduinoJSON, so I suspect my problem is how I'm using it.

It still crashes if I remove the clear() functions too, I just added them in to free up some memory.

The Code:
```
void ICACHE_FLASH_ATTR httpHandleInt() {
  if (webServer.hasArg("plain") == false) {
    httpSendIntError("No POST body data.");
    return;
  }

  if (strlen(webServer.arg("plain").c_str()) > OP_PROCESS_ARR_LIMIT) {
    httpSendIntError("POST body too large. Keep smaller than 512 bytes.");
    return;
  }

  // Block 1 Start
  DynamicJsonDocument JSONbuffer(512);
  deserializeJson(JSONbuffer, webServer.arg("plain"));

  JsonArray opList = JSONbuffer["ops"];
  char codeList[OP_PROCESS_ARR_LIMIT];
  int byteIndex = 4;

  for(JsonVariant elem: opList) {
    if (byteIndex > OP_PROCESS_ARR_LIMIT) {
      JSONbuffer.clear();
      char* errorMsg = strdup((const char*)F("Byte array length exceeds 512"));
      
      httpSendIntError(errorMsg); // Just outputs a JSON error message
      return;
    }
    if (elem.is<char*>()) {
      unsigned int res;
      char* inputChar = strdup(elem.as<char*>());
      boolean numRes = stringToNumber((int *)&res, inputChar);
      if (!numRes) {
        char* errorMsg = strdup((const char*)F("Invalid character in OP list"));
        httpSendIntError(byteIndex, inputChar, errorMsg);
        return;
      }
      codeList[byteIndex] = res;
      byteIndex++;
    } else if (elem.is<byte>()) {
      codeList[byteIndex] = elem.as<byte>();
      byteIndex++;
    } else if (elem.is<int>()) {
      int res = elem.as<int>();
      if (res > 0xFFFF) {
        char* errorMsg = strdup((const char*)F("Input number exceeds 0xFFFF (65535)"));
        httpSendIntError(byteIndex, (char *)elem.as<int>(), errorMsg);
        return;
      }
      codeList[byteIndex] = res & 0xFFFF;
      byteIndex++;
    } else {
      JSONbuffer.clear();
      StaticJsonDocument<512> JSONErrorBuffer;
      char JSONErrorMessageBuffer[512];
      
      JsonObject JSONErrorEncoder = JSONErrorBuffer.to<JsonObject>();
      JSONErrorEncoder[F("payload")];
      JSONErrorEncoder[F("payload")][F("meta")];
      JSONErrorEncoder[F("payload")][F("status")];
    
      JSONErrorEncoder[F("payload")][F("meta")][F("uptime")] = millis();
      JSONErrorEncoder[F("payload")][F("meta")][F("heapAvailable")] = heapAvailable();
      JSONErrorEncoder[F("payload")][F("meta")][F("stackAvailable")] = stackAvailable();
    
      JSONErrorEncoder[F("payload")][F("status")][F("code")] = F("400");
      JSONErrorEncoder[F("payload")][F("status")][F("codeMessage")] = F("Bad Request");
      JSONErrorEncoder[F("payload")][F("status")][F("errorDescription")] = F("OPs array does not only contain String<byte> or int types.");
      JSONErrorEncoder[F("payload")][F("status")][F("goodExample")] = F("{'ops': ['0x04', '0x05', 5, 7]}");
      JSONErrorEncoder[F("payload")][F("status")][F("badExample")] = F("{'ops': [0x04, 0x05, A, B, [1, 2], {'some': 'obj'} ]}");

      serializeJson(JSONErrorEncoder, JSONErrorMessageBuffer);
      
      webServer.send(400, "application/json", JSONErrorMessageBuffer);
      return;
    }
  }

  JSONbuffer.clear();
  // Block 1 End

  // Store the string result of each of the commands into an array of strings.
  char **execRes = (char **) calloc(byteIndex, sizeof(char*));
  char **echoBack = (char **) calloc(byteIndex, sizeof(char*));
  
  char errorMessage[32] = {0};
  // Basically loops and parses the inputs into commands to execute on the EEPROM memory
  processCommandInputHttp(codeList, byteIndex, execRes, errorMessage, echoBack); // Function is in serial_http_prog.ino
  
  if (strlen(errorMessage) > 0) {
    httpSendIntError(errorMessage);
    return;
  }

  // Block 2 start
  StaticJsonDocument<512> JSONResponseBuffer;
  char JSONResponseMessageBuffer[512];
  
  JsonObject JSONResponseEncoder = JSONResponseBuffer.to<JsonObject>();
  JSONResponseEncoder[F("payload")];
  JSONResponseEncoder[F("payload")][F("meta")];
  JSONResponseEncoder[F("payload")][F("status")];

  JSONResponseEncoder[F("payload")][F("meta")][F("uptime")] = millis();
  JSONResponseEncoder[F("payload")][F("meta")][F("heapAvailable")] = heapAvailable();
  JSONResponseEncoder[F("payload")][F("meta")][F("stackAvailable")] = stackAvailable();

  JSONResponseEncoder[F("payload")][F("status")][F("code")] = F("200");
  JSONResponseEncoder[F("payload")][F("status")][F("execRes[0]")] = execRes[0]; // Test output of the first result.

  serializeJson(JSONResponseEncoder, JSONResponseMessageBuffer);
  // Block 2 End

  delay(100); // Give power time to stablise to avoid low power crashes
  
  webServer.send(200, "application/json", JSONResponseMessageBuffer);
  
  for (int i = 0; i < OP_PROCESS_ARR_LIMIT && i < byteIndex; i++) {
    if (execRes[i]) {
      free(execRes[i]);
    }

    if (echoBack[i]) {
      free(echoBack[i]);
    }
  }
  
  if (execRes) {
    free(execRes);
  }
  if (echoBack) {
    free(echoBack);
  }
  
  delay(50);
}
```

ESP8266 Exceptions: https://links2004.github.io/Arduino/dc/deb/md_esp8266_doc_exception_causes.html

bblanchon, 2019-06-06:
Hi @Slyke,

Please apologize for the late answer.
I don't have time for a complete analysis of the code, so here are just the salient points.

### 1. Too much stack memory

The stack is limited to 4KB on the ESP8266, and most of it is already used by the framework.

This program puts several big variables in the stack:

```c++
char codeList[OP_PROCESS_ARR_LIMIT];
StaticJsonDocument<512> JSONResponseBuffer;
char JSONResponseMessageBuffer[512];
```

To avoid a stack overflow, use the heap instead.

### 2. Flash address treated as RAM address

The following line uses a Flash address as a RAM address:

```c++
char* errorMsg = strdup((const char*)F("Byte array length exceeds 512"));
```

Use `String` to easily move a Flash string to RAM.

### 3. Memory leak?

Are you freeing the memory from strdup() in the following call?
If not, you have a memory leak.

```c++
httpSendIntError(errorMsg);
```

You would not have this problem with `String`.

### Conclusion

Only one issue like that is sufficient to crash your program.
Fix the three and you should be good 👍 

BTW, I invite you to check out "The Missing C++ Course" of [Mastering ArduinoJson](https://arduinojson.org/book/), it covers all the topics involved here.

Best Regards,
Benoit

Slyke, 2019-06-07:
Hey @bblanchon thanks! Totally understandable. 

1. Changed them all to use the heap instead of the stack.
2. In the process of changing these to String see if it helps. Since it's a const char being passed into `strdup`, I didn't think it would cause a problem where it was coming from.
3. Yes, you're right. I wasn't clearing my memory (I didn't know that using `strdup` wouldn't deallocate the char * on destruction of function), I was wondering where my memory leak was from watching the heap size get smaller over time.

It's still crashing, but I think I will cut just this function out, run it in a separate project and see if I can track down the exact issue.

I also just purchased your book since I think you do an awesome job here. I'll have a read through it too!

closed 2019-06-07 by Slyke

bblanchon, 2019-06-07:
Hi @Slyke,

I'm glad I could help.
Thank you very much for purchasing the book, I hope you'll like it.

Best Regards,
Benoit

Slyke, 2019-06-13:
Just an update on this: After reading your book and trying a few things I've come to the conclusion that my stack and heap were crashing into each other. That's what was causing the random exception crashes. Thanks again for your help!