bblanchon/ArduinoJson #1022: Issue getting data from wunderground...

JasonMeredith1, 2019-06-10:
I am trying to get just a few pieces of data from the wunderground API...  I have been following along on several examples but I am missing where the returned string gets passed to deserialize.  can anyone here give me a clue what I need to add.

`/**
   BasicHTTPSClient.ino
    Created on: 20.08.2018
*/
#include <ArduinoJson.h>
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClientSecureBearSSL.h>

// Current fingerprint for https://api.wunderground.com
const uint8_t fingerprint[20] = {0x5D, 0xC3, 0x85, 0xD6, 0x19, 0x22, 0x99, 0x2B, 0x7B, 0xBB, 0x87, 0x8F, 0xD6, 0x9E, 0xF2, 0xD5, 0x44, 0xDA, 0x32, 0x28};

ESP8266WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("-------", "-------");
}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {
    std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);
    client->setFingerprint(fingerprint);
    HTTPClient https;
    Serial.print("[HTTPS] begin...\n");
    if (https.begin(*client, "https://api.weather.com/v3/wx/forecast/daily/5day?postalKey=78249:US&units=e&language=en-US&format=json&apiKey=-----REDACTED-----")) {
      Serial.print("[HTTPS] GET...\n");
      int httpCode = https.GET();
      if (httpCode > 0) {
        Serial.printf("[HTTPS] GET... code: %d\n", httpCode);
    if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
      Serial.printf("[HTTPS] httpCode:  %d\n" , httpCode);
      const size_t capacity = JSON_ARRAY_SIZE(1) + 20*JSON_ARRAY_SIZE(6) + 27*JSON_ARRAY_SIZE(12) + JSON_OBJECT_SIZE(21) + JSON_OBJECT_SIZE(27);
      Serial.printf("[JSON] capacity: %d\n" , capacity);
      DynamicJsonDocument doc(capacity);
      deserializeJson(doc, client);
      //DeserializationError error = deserializeJson(doc, client);
      //if (error) {
      //  Serial.print(F("deserializeJson() failed: "));
      //  Serial.println(error.c_str());
      //  return;
      Serial.print(doc["dayOfWeek"][0].as<char*>());  
       }
      } else {
        Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());
      }
 
      https.end();
    } else {
      Serial.printf("[HTTPS] Unable to connect\n");
    }
  }
  // Extract values
 // Serial.print(doc["dayOfWeek"][0].as<char*>());  
  Serial.println("Wait 180s before next round...");
  delay(180000);
}

`

bblanchon, 2019-06-12:
Hi @JasonMeredith1,

You can get the underlying `WiFiClient` by calling `HTTPClient::getStream()`:

```c++
DeserializationError error = deserializeJson(doc, client->getStream());
```

If you want a complete example using ESP8266HTTPClient, please check out the "GitHub" case study in [Mastering ArduinoJson](https://arduinojson.org/book/).

BTW, I thought Weather Underground shut down their API; how can you still be using it?

Best Regards,
Benoit

JasonMeredith1, 2019-06-12:
I got it figured out after a few hours banging around in the examples :), If you have a PWS (personal weather station) you get an API key from Wunderground.  I am now trying to figure out how to pass my returned string of data to an LED matrix so I can have a ticker with the weather for out flight communications center. with the Wemos Minis being so cheap I can thing a several projects to get on the workbench!  Thanks again for making such a great library!

Fair WInds,
-Jason

closed 2019-06-12 by JasonMeredith1