bblanchon/ArduinoJson #1023: How do I get a copy of a JsonDocument?

qwykx, 2019-06-12:
Hello

I am currently migrating my application from ArduinoJson 5 to AruduinoJson 6 and I would like to work with JsonObjects in my functions and therefore: How can I return a JsonObject form a JsonDocument within a function? In the following example the problem is that the JsonObject is a reference only and that means as soon as the the function returns the JsonDocument is deallocated if I understood the documentation right. So how can I use the JsonObject now without making the JsonDocument global?

`// ArduinoJson - arduinojson.org
// Copyright Benoit Blanchon 2014-2019
// MIT License
//
// This example shows how to deserialize a JSON document with ArduinoJson.

#include <iostream>
#include "ArduinoJson.h"

JsonObject getJson () {
      DynamicJsonDocument doc(300);
        
       // get the json via http here hard coded
      char json[] =
      "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
      
      // Deserialize the JSON document
      deserializeJson(doc, json);
      
      return doc.as<JsonObject>();
  }
  
  

int main() {
    
    JsonObject test = getJson();

  // Fetch values.
  //
  // Most of the time, you can rely on the implicit casts.
  // In other case, you can do doc["time"].as<long>();
  const char* sensor = test["sensor"];
  long time = test["time"];
  double latitude = test["data"][0];
  double longitude = test["data"][1];
  int size = test["data"].size();
 
 // Print values.
  std::cout << sensor << std::endl;
  std::cout << time << std::endl;
  std::cout << latitude << std::endl;
  std::cout << longitude << std::endl;
  std::cout << size << std::endl;

  return 0;
}
`

bblanchon, 2019-06-13:
Hi @qwykx,

Unlike `JsonObject`, `JsonDocument` has value semantics. When you assign a `JsonDocument` to another, it makes a deep copy.

Here is how you can fix your program:

```c++
DynamicJsonDocument getJson () {
  DynamicJsonDocument doc(300);

  // get the json via http here hard coded
  char json[] =
    "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

  // Deserialize the JSON document
  deserializeJson(doc, json);

  return doc;
}
```

Bonus: the capacity of the returned `DynamicJsonDocument` is computed from the actual memory usage of `doc`, so it can be significantly lower.

PS: I added more detail in the [documentation of `DynamicJsonDocument`](https://arduinojson.org/v6/api/dynamicjsondocument/).

Best Regards,
Benoit

brettbeeson, 2019-06-14:
Salut Benoit, et merci pour ArduinoJson!

I have a related question: I want to return and combine a JsonDocument. Specifically, I need to create:
```
   {
   "event":"settings",
   "data":{
           "saveFolder":"f",
           "freeHeap":190500
          }
   }
```
from a function which returns a DynamicJsonDocument with:
```
{
      "saveFolder":"f",
       "freeHeap":190500
}
```
I _think_ I have the right method. Complete sketch:
```
#include <ArduinoJson.h>
void merge(JsonObject& dest, const JsonObject& src) {
  for (JsonPair kvp : src) {
    dest[kvp.key()] = kvp.value();
  }
}

DynamicJsonDocument functionFromAnotherLand() {
  const char*inner = "{\"saveFolder\":\"sdcard\",\"freeHeap\":190500}";

  DynamicJsonDocument doc(JSON_OBJECT_SIZE(2) + 30);
  deserializeJson(doc, inner);
  return doc;
}

void setup() {
  Serial.begin(115200); while (!Serial);
}

void loop() {

  String json;

  // Containing Document. "data" will be completed by the function.
  DynamicJsonDocument combined(JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(2) + 30);
  combined["event"] = "settings";
  JsonObject dataObj = combined.createNestedObject("data");
  
  // Copy the document, automatically getting the right amount of memory
  DynamicJsonDocument inner = functionFromAnotherLand();

  // "Insert" the returned object into the Containing Document.
  merge(dataObj, inner.as<JsonObject>());
  
  serializeJson(combined, json);
  Serial.printf("MEM %d JSON: %s\n", ESP.getFreeHeap(),json.c_str());
  delay(1000);
}
```
Can you comment / confirm this is the correct way to use your library? I've adapted answers from your other responses.

Merci, Brett.

qwykx, 2019-06-21:
Hello @bblanchon 

Thank you very much for the clarification :) Would be nice to add that information to the migration document from version 5 to 6 because I haven't read through all the documentation stuff in version 6 yet.

Appreciate your library 👍 

Cheers mate

closed 2019-06-21 by qwykx

bblanchon, 2019-06-21:
You're welcome @qwykx; I added a paragraph about copying.

@brettbeeson, you should be able to do:

```c++
combined["data"] = functionFromAnotherLand();
```

If not, please open a new issue.

brettbeeson, 2019-06-23:
Thanks @bblanchon, that works

I had to add a casty bit, hence:

`combined["data"] = functionFromAnotherLand().as<JsonObject>();`