bblanchon/ArduinoJson #1024: Code is working on Wandbox but not working on Hardware

apanasara, 2019-06-13:
I am using NodeMCU 1.0 & 0.9 with PlatformIO.

After migrating to ArduinoJson v6.11.0, I am facing problem for parsing nesting Arrays.

> Below Same code is working on Wandbox but not working on Hardware

**Code on Arduino** _[WandBox](https://wandbox.org/)_
```cpp

#include <iostream>
#include "ArduinoJson.h"

int main() {
  
 char array[] = "[[1,2],[2,3]]";

DynamicJsonDocument doc(1024);

 if (error) {
    std::cerr << "deserializeJson() failed: " << error.c_str() << std::endl;
    return 1;
  }

  // Print values.
  std::cout << doc<< std::endl; // This prints: [[1,2],[2,3]]
  std::cout << doc[0] << std::endl; // This prints: [1,2]

  return 0;
}

```

**Code on Arduino** _PlatfiormIO NodeMCU_
```cpp
char array[] = "[[1,2],[2,3]]";

DynamicJsonDocument doc(1024);
deserializeJson(doc, array);

serializeJson(doc, Serial);
// This prints:
// "[[1,2],[2,3]]"

serializeJson(doc[0], Serial);
// This prints:
// null

JsonArray data = doc.as<JsonArray>();
serializeJson(data, Serial);
// This prints:
// null
```

apanasara, 2019-06-14:
I read documentation of v6 in which I found that ESP8266 is not in list of [supported hardware](https://arduinojson.org/hardware/)

So I will degrade library from v6 to back in v5 (v5 tested & supporting to ESP8266)

bblanchon, 2019-06-14:
Hi @apanasara,

ArduinoJson v6 supports ESP8266, and the program above runs as expected on my Adafruit Huzzah.

[image]

Please double check.

Best Regards,
Benoit

apanasara, 2019-06-14:
Nested arrays used

**Compilation message**
nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)

**Screenshot**
[image]

apanasara, 2019-06-15:
**Error-Observation**
[image]

I tried following mechanism to deserialize & found following solution. (don't know exact reason, you can share if know)

**Code Block**
```cpp
int Response(char *topic, const char *msg_buf)
{
  Serial.print("in Response Block : ");
  Serial.println(msg_buf);
  StaticJsonDocument<2048> req;
  DeserializationError err = deserializeJson(req, msg_buf, DeserializationOption::NestingLimit(4));

/*------following re-deserilization worked for me------*/
  err = deserializeJson(req, req.as<const char*>(), DeserializationOption::NestingLimit(4));
```

**Output**
[image]

bblanchon, 2019-06-20:
Hi @apanasara,

I'm gald you found a solution.
This article may be relevant: [Why is the input modified?](https://arduinojson.org/v6/faq/why-is-the-input-modified/)

Best Regards,
Benoit

closed 2019-06-20 by bblanchon