bblanchon/ArduinoJson #1025: Defining that DynamicJsonDocument is full

Elmaranin, 2019-06-14:
Hello, just a little question. Is there a way to define that DynamicJsonDocument is full and new data fields wouldn't be added? I'm using ArudinoJson in my own lib and I want my lib tells outside something like "insufficient memory for serialization" in order to library-users know that they must increase it (it is passed to library as parameter).
At this point there are no errors and I can't be sure that whole data I want is serialized. It would be great if there will be some parameter like "addedSize" that would return the whole size that user adds in document(even if it's not really added). In this case before serialization I can check if this parameter is greater that allocated memory - than some data are missed.

bblanchon, 2019-06-14:
Hi @Elmaranin,

[JsonObject::set()](https://arduinojson.org/v6/api/jsonobject/set/) and [JsonArray::add()](https://arduinojson.org/v6/api/jsonarray/add/) return a boolean that tells if the insertion succeeded.

Alternatively, you can compare [JsonDocument::memoryUsage()](https://arduinojson.org/v6/api/jsondocument/memoryusage/) with JsonDocument::capacity().

Best Regards,
Benoit

Elmaranin, 2019-06-14:
Thanks for response and for awesome library! 
However I don't get how one can use 'capacity' and 'memoryUsage' to reach my goal. Or you mean that I can check for free space before I'm gonna add new data with known size?

Elmaranin, 2019-06-14:
Here is example of my lib usage:
```
myLib->addJsonWithResponseHandler("fs-info", std::bind([this](JsonObject& payload, JsonObject& response) {
	FSInfo data = _fsCore->info();
	response["totalBytes"] = data.totalBytes;
	response["usedBytes"] = data.usedBytes;
	response["blockSize"] = data.blockSize;
	response["pageSize"] = data.pageSize;
	response["maxOpenFiles"] = data.maxOpenFiles;
	response["maxPathLength"] = data.maxPathLength;
	return true;
}, std::placeholders::_1, std::placeholders::_2));
```

After using **set** it would be something like this:
```
myLib->addJsonWithResponseHandler("fs-info2", std::bind([this](JsonObject& payload, JsonObject& response) {
	FSInfo data = _fsCore->info();
	StaticJsonDocument<200> tempJson;
	JsonObject tempJsonRef = tempJson.to<JsonObject>();
	tempJsonRef["totalBytes"] = data.totalBytes;
	bool res = response.set(tempJsonRef);
	if (!res) {
	}
	tempJson.clear();

	tempJsonRef["usedBytes"] = data.usedBytes;
	res = response.set(tempJsonRef);
	if (!res) {
	}
	tempJson.clear();

	tempJsonRef["blockSize"] = data.blockSize;
	res = response.set(tempJsonRef);
	if (!res) {
	}
	tempJson.clear();

	tempJsonRef["pageSize"] = data.pageSize;
	res = response.set(tempJsonRef);
	if (!res) {
	}
	tempJson.clear();

	tempJsonRef["maxOpenFiles"] = data.maxOpenFiles;
	res = response.set(tempJsonRef);
	if (!res) {
	}
	tempJson.clear();

	// response["maxPathLength"] = data.maxPathLength;
	return true;
}, std::placeholders::_1, std::placeholders::_2), 1000, 80);
```
Or something similar checking available space. Code looses it's beauty and becomes some messy. That's why solution with one-time checking before serialization looks great.

Another question: I didn't find any example of **set** method usage, but one I specified above works wrong - it doesn't place actual data into **response** object, it removes it once **tempJson.clear()** occur. I thought that **set** method copies actual data and source **tempJson** document can be cleared after that.

bblanchon, 2019-06-15:
Hi @Elmaranin,

I'm sorry, I mixed up ArduinoJson 5 and 6.
I was thinking of [JsonVariant:set()](https://arduinojson.org/v6/api/jsonvariant/set/).
I added an example on the page.

In your case, it would be something along those lines:

```c++
return response["totalBytes"].set(data.totalBytes) 
    && response["usedBytes"].set(data.usedBytes)
    && ...
```

Best Regards,
Benoit

Elmaranin, 2019-06-15:
Thank you very much for your answer and your great work!

closed 2019-06-15 by Elmaranin