bblanchon/ArduinoJson #1027: DynamicJsonDocument as/to<JsonObject> works wrong

Elmaranin, 2019-06-17:
Another question: In documentation for [JsonObject](https://arduinojson.org/v6/api/jsonobject/) there is example with **StaticJsonDocument** and I thought that everything must work just the same with **DynamicJsonDocument** but it's not.
This code prints "null":
```
	DynamicJsonDocument response(1024);
	JsonObject responseReference = response.to<JsonObject>();
	responseReference["test"] = "6";
	Serial.println(responseReference["test"].as<String>());
```
This code returns same "null":
```
	DynamicJsonDocument response(1024);
	response["test"] = "6";
	JsonObject responseReference = response.to<JsonObject>();
	Serial.println(responseReference["test"].as<String>());
```

The only workable way is to explicitly init document with some value and then use **as<JsonObject>**:
```
	DynamicJsonDocument response(1024);
	response["test"] = "6";
	JsonObject responseReference = response.as<JsonObject>();
	Serial.println(responseReference["test"].as<String>());
```

Elmaranin, 2019-06-17:
Maybe it is the right way to use **JsonObject** with **DynamicJsonDocument** but it must be in documentation then =).

bblanchon, 2019-06-17:
Hi @Elmaranin,

The first example prints "6" as expected, see demo: https://wandbox.org/permlink/lL9OO9mgGwNcsiMj

The second example prints "null" as expected, because [JsonDocument::to<T>()](https://arduinojson.org/v6/api/jsondocument/to/) clears the document.

The third example prints "6" as expected, because [JsonDocument::as<T>()](https://arduinojson.org/v6/api/jsondocument/as/) doesn't clear the document.

Please tell me exactly where I should change the documentation and I'll be happy to do so.

Best Regards,
Benoit

Elmaranin, 2019-06-18:
I don't get it, but yesterday I did about 15 tests before I wrote this issue. Today I can't reproduce it at all. Sorry to bother you.

closed 2019-06-18 by Elmaranin

bblanchon, 2019-06-19:
No worries @Elmaranin.
Thanks again for using ArduinoJson.