bblanchon/ArduinoJson #1028: ArduinoJson 6. When use DynamicJsonDocument get "Not enough memory"

Warlib1975, 2019-06-18:
Hello,
When I use DynamicJsonDocument with specified capacity
DynamicJsonDocument doc(capacity);

when customer make JSON file more than specified in the capacity I get error: "Not enough memory". Corresponding https://arduinojson.org/v6/api/dynamicjsondocument/ I see the phrase "If this case, doc2 keeps its capacity of 2048 bytes, except if 2048 is not enough, in which case doc2 allocates a larger memory pool.", so if the DynamicJsonDocument don't enough memory it must allocate more memory, but it doesn't work. 

I don't need high performances during JSON processing, but a customer can create 5000 bytes file or 50000 bytes file. How I should proper use DynamicJsonDocument?

Thank you.

bblanchon, 2019-06-18:
Hi @Warlib1975,

DynamicJsonDocument can reallocate its memory pool only when copying another JsonDocument.
In other situations, it has a fixed capacity.

If the DynamicJsonDocument is too small, you must increase its capacity.
If your program must support a JsonDocument of 50KB, you must allocate 50KB:

```c++
DynamicJsonDocument doc(50000);
```

If there is not enough memory in your microcontroller, then you cannot parse such a large input in one chunk.

Best Regards,
Benoit

Warlib1975, 2019-06-18:
> DynamicJsonDocument can reallocate its memory pool only when copying another JsonDocument.
> In other situations, it has a fixed capacity.

Thank you for the replay. But in this case, it seems to look like StaticJsonDocument. What's a difference,.

Which way is better to use for the case when a customer can use any size of JSON file as a config file? I need to import JSON config file and use minimum memory of ESP32 microcontroller.

bblanchon, 2019-06-19:
Hi @Warlib1975 

Yes, `DynamicJsonDocument` and [`StaticJsonDocument`](https://arduinojson.org/v6/api/staticjsondocument/) are very similar; they only differ by the memory location:

* [`StaticJsonDocument`](https://arduinojson.org/v6/api/staticjsondocument/) allocates in the stack (most likely)
* [`DynamicJsonDocument`](https://arduinojson.org/v6/api/dynamicjsondocument/) allocates in the heap

In your case, you must use a [`DynamicJsonDocument`](https://arduinojson.org/v6/api/dynamicjsondocument/) because your document cannot fit in the stack of an ESP32.

"use any size of JSON file as a config file?" is impossible, by definition.
"use minimum memory" is in contradiction with your other goal.
ArduinoJson is rigid in its memory allocation, but it helps you write safe and efficient code.

Best Regards,
Benoit

closed 2019-07-20 by bblanchon