bblanchon/ArduinoJson #1031: Concatenate char arrays -> null in serialized document

Tuebel, 2019-06-23:
When I try to serialize a concatenated char arrays or strings, I get null entries in the resulting serialized JSON. Here is a small example:

```cpp
  char test_1[] = "test_1";
  char test_2[20];
  strcpy(test_2, "test_2");
  char test_3[20];
  strcpy(test_3, "test");
  strcat(test_3, "_3");
  
  const size_t capacity = JSON_OBJECT_SIZE(4);
  DynamicJsonDocument doc(capacity);
  doc["test_1"] = test_1;
  doc["test_2"] = test_2;
  doc["test_3"] = test_3;
  serializeJson(doc, Serial);
  Serial.println();
  Serial.println(test_3);
  doc["test_3"] = (const char*)test_3;
  serializeJson(doc, Serial);
```

This results in the following output:
```
{"test_1":"test_1","test_2":"test_2","test_3":null}
test_3
{"test_1":"test_1","test_2":"test_2","test_3":"test_3"}
```

The same happens if I try to concatenate `String`s via `+`.

I found a workaround: casting the `char[]` to `const char*` forces a copy and gives me the desired output. The same works for `String.c_str()`.

I use this on a Wemos D1 mini R2 board.
With best regards,
Tim

bblanchon, 2019-06-25:
Hi Tim,

You must increase the capacity of the `DynamicJsonDocument`, because it is too small to hold a copy of `test3`. Using a `const char*` prevents the copy, that's why it works.

See:

* [Why is the output incomplete?](https://arduinojson.org/v6/faq/why-is-the-output-incomplete/)

Best Regards,
Benoit

Tuebel, 2019-06-27:
Thanks for your quick reply Benolt!
I confused when a copy is created and when not. Also the larger story solved the problem :)

Best regards,
Tim

closed 2019-06-27 by Tuebel

bblanchon, 2019-06-29:
You're welcome Tim.
Thank you for using ArduinoJson.