bblanchon/ArduinoJson #1034: Deserialise stored JSON into nested object?

mdelain, 2019-06-24:
Hello Anthony,

I'm getting used to ArduinoJson but for the moment I can't manage to import a JSON into one of the nested object of my Json Document.

For instance I need to create a MQTT message that could contain more or less members depeinding on the case:
```
{
    "data":{
        "temperature":{
            "t":"int32",
            "v":23
        }
    }
}
```
or
```
{
    "data":{
        "temperature":{
            "t":"int32",
            "v":23
        },
        "hygrometry":{
            "t":"int32",
            "v":35
        }
    }
}
```

I'm trying to store the `temperature` and `hygrometry` objects in char arrays, and load them into the `data` object when needed. But I did not find how to do it :(

```
<ArduinoJson.h>

StaticJsonDocument<150> doc;
char json[] = "\"hygrometry\": { \"t\":\"int32\", \"v\": 35 }";

void setup() {
  Serial.begin(9600);
  while (!Serial);
}

void loop() {
  JsonObject data  = doc.createNestedObject("data");
  JsonObject temperature  = cfg.createNestedObject("temperature");
  deserializeJson(temperature, json);
  serializeJsonPretty(doc, Serial);
  while (1);
}
```
Thanks!

mdelain, 2019-06-29:
Finally I went with a workaround: first I do a `deserializeJson()` of the full JSON, then I use `JsonObject::remove()` to remove unwanted objects.

I did not find any way to inject/deserialize JSON into nested objects other than deserializing my JSON buffer into a temporary object, then recopy members into the JSON document.

bblanchon, 2019-07-01:
Hi @mdelain,

I'm sorry I couldn't come with a solution to your problem.
Luckily you found one yourself.
Thank you for using ArduinoJson.

Best Regards,
Benoit

closed 2019-07-01 by bblanchon