bblanchon/ArduinoJson #1035: Compile error on particle argon board

tyczj, 2019-06-29:
When I try to compile with this library included in my particle argon project OS version 1.3.0 I get this error.

Error:

```
In file included from /Users/jefftycz/Documents/Projects/iot-ac/iot-ac/lib/ArduinoJson/src/ArduinoJson/Deserialization/deserialize.hpp:8:0,
                 from /Users/jefftycz/Documents/Projects/iot-ac/iot-ac/lib/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:7,
                 from /Users/jefftycz/Documents/Projects/iot-ac/iot-ac/lib/ArduinoJson/src/ArduinoJson.hpp:33,
                 from /Users/jefftycz/Documents/Projects/iot-ac/iot-ac/lib/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/jefftycz/Documents/Projects/iot-ac/iot-ac/src/iot-ac.ino:6:
/Users/jefftycz/Documents/Projects/iot-ac/iot-ac/lib/ArduinoJson/src/ArduinoJson/Deserialization/ArduinoStreamReader.hpp: In member function 'int ArduinoJson6110_00000::ArduinoStreamReader::read()':
/Users/jefftycz/Documents/Projects/iot-ac/iot-ac/lib/ArduinoJson/src/ArduinoJson/Deserialization/ArduinoStreamReader.hpp:22:35: error: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'char*' [-fpermissive]
     return _stream.readBytes(&c, 1) ? c : -1;
```

bblanchon, 2019-07-01:
Hi @tyczj,

For some reason, I couldn't compile with the particle command line, so I'm not sure my change really fixes the compilation for Argon.
Please test and confirm. Thanks!

Best Regards,
Benoit

closed 2019-07-01 by bblanchon

tyczj, 2019-07-01:
Seems to have fixed it, thanks

hengel2810, 2019-07-05:
The newest version in the Particle Web IDE is still 6.11.0 without this fix, any ideas how to get the new version with the fix within the Web IDE ?

bblanchon, 2019-07-08:
@hengel2810, I just published [ArduinoJson 6.11.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.11.2).
It should be available in the Web IDE now.

commit 7d40a541 (2019-07-01): Fix invalid conversion error on Particle Argon (closes #1035)