bblanchon/ArduinoJson #1036: Compile Error in src/ArduinoJson/Strings/SizedFlashStringAdapter.hpp

l-gothberg, 2019-07-02:
Installed Version: 6.11.1

Installed via Arduino library manager.  Version 5.13.5 seems to work fine.  Testing reveals that even declaring the `#include <ArduinoJson.h>` is enough to replicate error.

Error Message:
Line 72:  strncmp_P was not declared in this scope.
    return -strncmp_P(other, reinterpret_cast<const char*>(_str), _size);

bblanchon, 2019-07-02:
Hi @l-gothberg,

This error happens on platforms that define `PROGMEM` but don't provide the required functions like `strncmp_P()`.
To workaround this problem, you need to define [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/) to 1.

Best Regards,
Benoit

l-gothberg, 2019-07-03:
I added the `#define ARDUINOJSON_ENABLE_PROGMEM 1` to my code, but still get the same error.  Would it be anything related to the fact that I'm using a Particle Xenon instead of a standard Arduino?  Everything else seems to be working fine, if I can solve this issue.

Here's the full output of the compile error:
```> Executing task: make -f '/Users/leo/.particle/toolchains/buildscripts/1.5.1/Makefile' compile-user -s <

Creating /Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/target/module_user_memory.ld ...
In file included from /Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/lib/ArduinoJson/src/ArduinoJson/Strings/StringAdapters.hpp:35:0,
                 from /Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/lib/ArduinoJson/src/ArduinoJson/Misc/SerializedValue.hpp:7,
                 from /Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/lib/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:7,
                 from /Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/lib/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:7,
                 from /Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/lib/ArduinoJson/src/ArduinoJson.hpp:17,
                 from /Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/lib/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/src/xenon-desk-controls.ino:7:
/Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/lib/ArduinoJson/src/ArduinoJson/Strings/SizedFlashStringAdapter.hpp: In member function 'int8_t ArduinoJson6111_11000::SizedFlashStringAdapter::compare(const char*) const':
/Users/leo/Desktop/github/home-mesh-devices/xenon-desk-controls/lib/ArduinoJson/src/ArduinoJson/Strings/SizedFlashStringAdapter.hpp:18:72: error: 'strncmp_P' was not declared in this scope
     return -strncmp_P(other, reinterpret_cast<const char*>(_str), _size);
                                                                        ^
make[3]: *** [../build/target/user/platform-14-m/xenon-desk-controls/src/xenon-desk-controls.o] Error 1
make[2]: *** [user] Error 2
make[1]: *** [modules/xenon/user-part] Error 2
make: *** [compile-user] Error 2
The terminal process terminated with exit code: 2

Press any key to close the terminal.```

bblanchon, 2019-07-03:
I'm sorry, I made a mistake in my previous message 🤦‍♂️
You must define it to `0` (and not `1`)

l-gothberg, 2019-07-03:
Ah!  Thanks!  All good now.

closed 2019-07-03 by l-gothberg