bblanchon/ArduinoJson #1037: Compile error in VariantData.hpp

m-andersen, 2019-07-02:
During compilation of ArduinoJson 6.9.0 I get the following compile error:

compiler version:
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0

compile error:
In file included from /home/nielses/workspace_cu_apps/libcu/src/PC/../../../libjson/src/ArduinoJson/Array/ArrayRef.hpp:7:0,
                 from /home/nielses/workspace_cu_apps/libcu/src/PC/../../../libjson/src/ArduinoJson.hpp:9,
                 from ../NetworkDiagnosticsController.h:19,
                 from ../NetworkDiagnosticsController.cpp:14:
/home/nielses/workspace_cu_apps/libcu/src/PC/../../../libjson/src/ArduinoJson/Array/../Variant/VariantData.hpp: In member function ‘void CNetworkDiagnosticsController::GetStatus(std::ostream&)’:
/home/nielses/workspace_cu_apps/libcu/src/PC/../../../libjson/src/ArduinoJson/Array/../Variant/VariantData.hpp:338:12: error: ‘*((void*)& jsonDocument +24)’ is used uninitialized in this function [-Werror=uninitialized]
     _flags &= KEY_IS_OWNED;
     ~~~~~~~^~~~~~~~~~~~~~~

Looks like compiler doesn't like POD class.

bblanchon, 2019-07-02:
Hi @m-andersen,

ArduinoJson 6.11.1 builds fine on GCC 7.4.0:
https://travis-ci.org/bblanchon/ArduinoJson/jobs/552910370
Can you upgrade?

Best Regards,
Benoit

m-andersen, 2019-07-02:
Hi

Thanks, I will test it out and let you know.
What compile flags do you use? It might be if I have aggressive flags set that it outputs errors. Usually we treat warnings as errors. From the link you sent I don't see many flags.

br
Martin

bblanchon, 2019-07-03:
Hi Martin,

On GCC 7.4, it should be:

```
-pedantic
-std=c++98
-Wall
-Wcast-align
-Wcast-qual
-Wconversion
-Wctor-dtor-privacy
-Wdisabled-optimization
-Werror
-Wextra
-Wformat=2
-Winit-self
-Wlogical-op
-Wmissing-include-dirs
-Wno-vla
-Wnoexcept
-Wnon-virtual-dtor
-Wold-style-cast
-Woverloaded-virtual
-Wparentheses
-Wredundant-decls
-Wshadow
-Wsign-promo
-Wstrict-aliasing
-Wstrict-null-sentinel
-Wundef
```

See [test/CMakeLists.txt](https://github.com/bblanchon/ArduinoJson/blob/6.x/test/CMakeLists.txt).

After seeing your first message, I added `-Wuninitialized` but it didn't change anything so I didn't commit the change. ([the documentation](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html) confirms that this flag is already included with `-Wall`).

Best Regards,
Benoit

m-andersen, 2019-07-18:
Hi

People are on vacation. I will get back to you again soon regarding this.

Martin

m-andersen, 2019-07-30:
Hi

You can close this. We have different compiler flags and will take a look when we upgrade at some point.

br
Martin

closed 2019-07-30 by bblanchon