bblanchon/ArduinoJson #1040: Code is not compiling

notSoTechnical, 2019-07-04:
Arduino: 1.8.9 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

/Users/user/Documents/Arduino/libraries/arduinoWebSockets-2.1.4/src/WebSocketsClient.cpp: In member function 'void WebSocketsClient::loop()':
/Users/user/Documents/Arduino/libraries/arduinoWebSockets-2.1.4/src/WebSocketsClient.cpp:172:30: error: 'class WiFiClientSecure' has no member named 'setCACert'
                 _client.ssl->setCACert((const uint8_t *)_CA_cert, strlen(_CA_cert) + 1);

                              ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bblanchon, 2019-07-04:
Hi @notSoTechnical,

This is a question for arduinoWebSockets, not ArduinoJson, but you probably just need to update the ESP8266 core.

Best Regards,
Benoit

notSoTechnical, 2019-07-04:
You are right!  I got it fixed.  Thanks anyway!

closed 2019-07-04 by notSoTechnical