bblanchon/ArduinoJson #1041: assign json value to variable

AlexFundorin, 2019-07-06:
ArduinoJson 6 on vscode with platformio. Using is with ESP32.
I have a "config.json" file in SPIFFS partition.
```
{
    "testString1": "text",
    "testString2": "notext"
}
```

I also have a variable:
```
 char* testString[] = "false";
```
What I need is to assign "testString1"s "text" value to testString variable.

This is my code so far:
```
void flashSetup(){
    SPIFFS.begin();
    File configFile = SPIFFS.open("/config.json");
    StaticJsonDocument<512> doc;
    deserializeJson(doc, configFile);
    testString = doc["testString1"];
}
```

What would be the right code to be able to successfully assign that value to the variable?
For now I'm getting different types of errors, while trying to perform the assignment.
The most recent ones are these:
```
1. error: invalid conversion from 'char**' to 'long int' [-fpermissive]
```
```
2. error: incompatible types in assignment of 'ArduinoJson6111_00000::enable_if<true, ArduinoJson6111_00000::MemberProxy<ArduinoJson6111_00000::JsonDocument&, const char*> >::type {aka ArduinoJson6111_00000::MemberProxy<ArduinoJson6111_00000::JsonDocument&, const char*>}' to 'char* [6]'
```

bblanchon, 2019-07-08:
Hi @AlexFundorin,

You should change the type of `testString` to `const char*`.

See also:

* [FAQ: Error: invalid conversion from 'const char*' to 'char*' [-fpermissive]](https://arduinojson.org/v6/faq/error-invalid-conversion-from-const-char-to-char/)

Best Regards,
Benoit

AlexFundorin, 2019-07-08:
Thank you. The issue was that I was printing testString's pointer instead of it's value.

closed 2019-07-08 by AlexFundorin