bblanchon/ArduinoJson #1044: No match for operator

lodv, 2019-07-08:
Dear Benoit,

I try to compile the sonoff.io file under Arduino 1.8.8
I installed the ArduinoJson 6.11.1 version.
I got the following error message:

/home/lodv/Arduino/sonoff/support.ino: In function 'bool JsonTemplate(const char*)':
support:1084:61: error: no match for 'operator|' (operand types are 'ArduinoJson::JsonArraySubscript' and 'int')
       Settings.user_template.gp.io[i] = obj[D_JSON_GPIO][i] | 0;
                                                             ^
support:1088:37: error: no match for 'operator|' (operand types are 'ArduinoJson::JsonObjectSubscript<const char*>' and 'int')
     uint8_t flag = obj[D_JSON_FLAG] | 0;
                                     ^
Meerdere bibliotheken gevonden voor "esp-knx-ip.h"
Gebruikt: /home/lodv/Arduino/libraries/esp-knx-ip-0.5.1
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/esp-knx-ip-0.5.1
Meerdere bibliotheken gevonden voor "PubSubClient.h"
Gebruikt: /home/lodv/Arduino/libraries/pubsubclient
Niet gebruikt: /home/lodv/Arduino/libraries/PubSubClient-EspEasy-2.7.12
Niet gebruikt: /home/lodv/Arduino/libraries/PubSubClient
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/PubSubClient-EspEasy-2.6.09
Meerdere bibliotheken gevonden voor "ihx.h"
Gebruikt: /home/lodv/Arduino/libraries/C2Programmer-1.0.0
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/C2Programmer-1.0.0
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/C2Programmer-1.0.0
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/C2Programmer-1.0.0
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/C2Programmer-1.0.0
Meerdere bibliotheken gevonden voor "RCSwitch.h"
Gebruikt: /home/lodv/Arduino/libraries/rc-switch-2.6.2.13
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/rc_switch
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/rc-switch-2.6.2.13
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/rc_switch
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/rc-switch-2.6.2.13
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/rc_switch
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/rc-switch-2.6.2.13
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/rc_switch
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/rc-switch-2.6.2.13
Meerdere bibliotheken gevonden voor "NewPing.h"
Gebruikt: /home/lodv/Arduino/libraries/NewPing-1.9.1
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/NewPing-1.9.1
Meerdere bibliotheken gevonden voor "ArduinoJson.h"
Gebruikt: /home/lodv/Arduino/libraries/ArduinoJson
Niet gebruikt: /home/lodv/Arduino/libraries/arduino_315611
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/ArduinoJson-5.11.2
Meerdere bibliotheken gevonden voor "IRremoteESP8266.h"
Gebruikt: /home/lodv/Arduino/libraries/IRremoteESP8266
Niet gebruikt: /home/lodv/Arduino/libraries/IRremoteESP8266-2.6.0
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/IRremoteESP8266-2.5.2.03
Meerdere bibliotheken gevonden voor "TasmotaSerial.h"
Gebruikt: /home/lodv/Arduino/libraries/TasmotaSerial-2.3.1
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/TasmotaSerial-2.2.0
Meerdere bibliotheken gevonden voor "TasmotaModbus.h"
Gebruikt: /home/lodv/Arduino/libraries/TasmotaModbus-1.1.0
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/TasmotaModbus-1.1.0
Meerdere bibliotheken gevonden voor "NeoPixelBus.h"
Gebruikt: /home/lodv/Arduino/libraries/NeoPixelBus_by_Makuna
Niet gebruikt: /home/lodv/Arduino/libraries/NeoPixelBus-2.2.9
Niet gebruikt: /home/lodv/arduino-1.8.8/libraries/NeoPixelBus-2.2.9
exit status 1
no match for 'operator|' (operand types are 'ArduinoJson::JsonArraySubscript' and 'int')

Niet gebruikt means not used and Meerdere bibliotheken gevonden means found more libraries

Can you help me?
Thank you in advance.

Yvonne de Vries

bblanchon, 2019-07-09:
Hi @lodv,

`ArduinoJson::JsonArraySubscript` is a class from ArduinoJson 5.
There is probably another installation of ArduinoJson one your computer.

Best Regards,
Benoit

lodv, 2019-07-11:
Dear Benoit,

That's right, but I downgraded it to version 5 and that doesn't help.
I suppose that there is a conflict between the latest Sonoff version and your ArduinoJson.
Could that be the case?
Thank you.
Regards,

Yvonne

bblanchon, 2019-07-15:
Hi Yvonne,

It seems that Sonoff uses Arduino 5.13.4, please confirm with them.
Looking at your log, it looks like you have several versions of ArduinoJson:

```
/home/lodv/Arduino/libraries/ArduinoJson
/home/lodv/arduino-1.8.8/libraries/ArduinoJson-5.11.2
```

Can you check?

Best Regards,
Benoit

lodv, 2019-07-24:
Dear Benoit,

I deleted the first Arduino directory.
But is it correct that the Arduino-1.8.8 creates a new Arduino directory by itself? Although it has only 1 library directory and the readme.txt file inside?
I decided not to use the Sonoff zip file as it seems to be not compatible with the ArduinoJson version I use . I use some other files I found on the internet. I used the instruction movie from Superhouse and I updated the Sonoff without Arduino.
Thank you for your patience.

Kind regards,

Yvonne

bblanchon, 2019-07-26:
Hi Yvonne,

I'm not sure I understand your question.
It's OK to have a `Arduino/library` directory in your home, because that's where you install the libraries.

I'm glad you made it work, can you write the steps here so to help other users?

Best Regards,
Benoit

lodv, 2019-07-28:
Hi Benoit,

I decided to update my Sonoff with the ESPtool instead of the Arduino ide. 
So, the problem with the JSON file was bypassed. I understand that I had two JSON versions installed and that gave conflicts. But I also had the problem that the Sonoff software was written in JSON 5 and the latest Arduino software I had installed uses a later version of JSON. So I decided to go a different path and update the Sonoff software using:
* Sonoff bin file
* ESPtool to upload the Sonoff Tasmota software
I know I can install my own written code on the Sonoff using Arduino. 
So I think I solved the problem this way.
Hope I made myself clear and maybe somebody can take advantage of this.

Kind regards,

Yvonne

closed 2019-07-28 by lodv