bblanchon/ArduinoJson #1045: Error: using std::string

sticilface, 2019-07-09:
Looking at the documentation this should work `size_t serializeMsgPack(const JsonDocument& doc, std::string& output);`
Compiling on the ESP32 Dev env within arduino.  

```cpp
#include <string>
#include <ArduinoJson.h>

std::string jsonString; 
void setup() {
DynamicJsonDocument doc(2048);
serializeMsgPack(doc, jsonString);
}

void loop() {
}
```

but it fails giving this error
```
In file included from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:37:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/amelvin/Documents/Arduino/deleteme/deleteme.ino:2:
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackSerializer.hpp: In instantiation of 'void ArduinoJson691_000::MsgPackSerializer<TWriter>::writeByte(uint8_t) [with TWriter = std::__cxx11::basic_string<char>; uint8_t = unsigned char]':
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackSerializer.hpp:144:14:   required from 'void ArduinoJson691_000::MsgPackSerializer<TWriter>::visitNull() [with TWriter = std::__cxx11::basic_string<char>]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:16:5:   required from 'void ArduinoJson691_000::variantAccept(const ArduinoJson691_000::VariantData*, Visitor&) [with Visitor = ArduinoJson691_000::MsgPackSerializer<std::__cxx11::basic_string<char> >]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:347:18:   required from 'void ArduinoJson691_000::VariantConstRef::accept(Visitor&) const [with Visitor = ArduinoJson691_000::MsgPackSerializer<std::__cxx11::basic_string<char> >]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:21:39:   required from 'void ArduinoJson691_000::JsonDocument::accept(Visitor&) const [with Visitor = ArduinoJson691_000::MsgPackSerializer<std::__cxx11::basic_string<char> >]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Serialization/serialize.hpp:21:3:   required from 'typename ArduinoJson691_000::enable_if<(! ArduinoJson691_000::IsWriteableString<TPrint>::value), unsigned int>::type ArduinoJson691_000::serialize(const TSource&, TPrint&) [with TSerializer = ArduinoJson691_000::MsgPackSerializer; TSource = ArduinoJson691_000::BasicJsonDocument<ArduinoJson691_000::DefaultAllocator>; TPrint = std::__cxx11::basic_string<char>; typename ArduinoJson691_000::enable_if<(! ArduinoJson691_000::IsWriteableString<TPrint>::value), unsigned int>::type = unsigned int]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackSerializer.hpp:172:38:   required from 'size_t ArduinoJson691_000::serializeMsgPack(const TSource&, TDestination&) [with TSource = ArduinoJson691_000::BasicJsonDocument<ArduinoJson691_000::DefaultAllocator>; TDestination = std::__cxx11::basic_string<char>; size_t = unsigned int]'
/Users/amelvin/Documents/Arduino/deleteme/deleteme.ino:15:33:   required from here
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackSerializer.hpp:153:19: error: 'class std::__cxx11::basic_string<char>' has no member named 'write'
     _bytesWritten += _writer->write(c);
                   ^
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackSerializer.hpp: In instantiation of 'void ArduinoJson691_000::MsgPackSerializer<TWriter>::writeBytes(const uint8_t*, size_t) [with TWriter = std::__cxx11::basic_string<char>; uint8_t = unsigned char; size_t = unsigned int]':
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackSerializer.hpp:89:15:   required from 'void ArduinoJson691_000::MsgPackSerializer<TWriter>::visitString(const char*) [with TWriter = std::__cxx11::basic_string<char>]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:39:53:   required from 'void ArduinoJson691_000::VariantData::accept(Visitor&) const [with Visitor = ArduinoJson691_000::MsgPackSerializer<std::__cxx11::basic_string<char> >]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:14:5:   required from 'void ArduinoJson691_000::variantAccept(const ArduinoJson691_000::VariantData*, Visitor&) [with Visitor = ArduinoJson691_000::MsgPackSerializer<std::__cxx11::basic_string<char> >]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:347:18:   required from 'void ArduinoJson691_000::VariantConstRef::accept(Visitor&) const [with Visitor = ArduinoJson691_000::MsgPackSerializer<std::__cxx11::basic_string<char> >]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:21:39:   required from 'void ArduinoJson691_000::JsonDocument::accept(Visitor&) const [with Visitor = ArduinoJson691_000::MsgPackSerializer<std::__cxx11::basic_string<char> >]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Serialization/serialize.hpp:21:3:   required from 'typename ArduinoJson691_000::enable_if<(! ArduinoJson691_000::IsWriteableString<TPrint>::value), unsigned int>::type ArduinoJson691_000::serialize(const TSource&, TPrint&) [with TSerializer = ArduinoJson691_000::MsgPackSerializer; TSource = ArduinoJson691_000::BasicJsonDocument<ArduinoJson691_000::DefaultAllocator>; TPrint = std::__cxx11::basic_string<char>; typename ArduinoJson691_000::enable_if<(! ArduinoJson691_000::IsWriteableString<TPrint>::value), unsigned int>::type = unsigned int]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackSerializer.hpp:172:38:   required from 'size_t ArduinoJson691_000::serializeMsgPack(const TSource&, TDestination&) [with TSource = ArduinoJson691_000::BasicJsonDocument<ArduinoJson691_000::DefaultAllocator>; TDestination = std::__cxx11::basic_string<char>; size_t = unsigned int]'
/Users/amelvin/Documents/Arduino/deleteme/deleteme.ino:15:33:   required from here
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackSerializer.hpp:157:19: error: 'class std::__cxx11::basic_string<char>' has no member named 'write'
     _bytesWritten += _writer->write(p, n);
```

my actual goal is to get this to work
```cpp
using PsramString = std::basic_string<char, std::char_traits<char>,spram_allocator<char>>; 
```

so I can use the PSRAM allocator and get all the strings onto the external ram.  
However, I got stuck as just trying to use `std::string`

sticilface, 2019-07-09:
# FAIL

`ARDUINOJSON_ENABLE_STD_STRING =  1`

for those who don't read the API reference first.

sticilface, 2019-07-09:
Actually... not so fast! 

Support for `std::string` exists but that is only `std::basic_string<char>`  crucially which only uses default allocator and i'd like to dump a whole set of strings on the psram which i can do using my own allocator.... 

```cpp
using PsramString = std::basic_string<char, std::char_traits<char>,spram_allocator<char>>; 
```

Can std::string be extended to allow such uses?

bblanchon, 2019-07-09:
*Answer to your first message:*
Unfortunately, there is no reliable way to tell if `std::string` available, so ArduinoJson tries to guess from the `ARDUINO` macro.

* If `ARDUINO` is defined, it assumes that `String` is available but not `std::string`.
* If `ARDUINO` is undefined, it assumes that `std::string` is available but not `String`.

*Answer to your last message:*
You're right: it only supports `std::basic_string<char>`.
Let's turn this into a feature request.

sticilface, 2019-07-09:
That would be amazing thank you.

closed 2019-08-12 by bblanchon

bblanchon, 2019-08-12:
Fix available in [ArduinoJson 6.11.4](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.11.4)

commit 1e9cc285 (2019-08-12): Added support for `basic_string<char, traits, allocator>` (closes #1045)