bblanchon/ArduinoJson #1046: Trouble serialising and deserialising MsgPack using Arduino String

sticilface, 2019-07-11:
I'm struggling to work with messagePack.  I can get my code to work with Json encoding and decoding... I essentially take a document (only one key ) created with JSON, and want to encode it into msgpack to pass around as it is more efficient. 

I've written this minimal example:

```cpp
#include <ArduinoJson.h>

String input = "{\"rgbww\":[0,0,0,0,255],\"time\":100,\"brightness\":255}";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  DynamicJsonDocument data(500);

  DeserializationError err = deserializeJson(data, input); 
  if (!err)
  {
    Serial.println("Json deserialised: ");
    serializeJsonPretty(data,Serial);
    Serial.println(); 

    String msgpack;
    size_t length = measureMsgPack(data); 
    msgpack.reserve(length + 1);  //  take out this line

    if (serializeMsgPack(data, msgpack)) {
      Serial.printf("MsgPack serialised: %s, stringlen = %u, actuallen = %u\n", msgpack.c_str(), msgpack.length(), length );
      DynamicJsonDocument data2(500);
      DeserializationError err2 = deserializeMsgPack(data2, msgpack);

      Serial.printf("MsgPack deserialise: %s\n", err2.c_str() ) ;

    }

  } else {
    Serial.printf("Json deserialised ERROR: %s\n", err.c_str() );
  }

}
```
`  msgpack.reserve(length + 1);  //  take out this line`
without this line the string is 29 char long, and the decode fails.  `MsgPack deserialise: IncompleteInput`
reserving the string length + 1 creates equal length strings and the decode works

any idea why?

sticilface, 2019-07-11:
After a fairly long debug process i'm not certain the Arduino String is compatible with msgpack.  

Difficult to prove at the moment, but my sketch involves copying the string into a lambda function for use later and the copy does not seem to complete when i use Arduino String.  if I switch to std::string then it copies fine and works.  

Below is the string hex output pre_lambda and  captured value  for `std::string`
```
Decoding Pre: Cmd: É•rgbwwïÃä+Ã‚, stringlen = 36, actual len = 36
00000000: 83 A5 72 67 62 77 77 95 CC 8A 2B CC E2 00 00 A4  É•rgbwwïÃä+Ã‚..§
00000010: 74 69 6D 65 CD 01 2C AA 62 72 69 67 68 74 6E 65  timeÕ.,™brightne
00000020: 73 73 CC FF                                      ssÃˇ

Decoding Capture: Cmd: É•rgbwwï
00000000: 83 A5 72 67 62 77 77 95 00 00 00 00 CC FF A4 74  É•rgbwwï....Ãˇ§t
00000010: 69 6D 65 64 AA 62 72 69 67 68 74 6E 65 73 73 CC  imed™brightnessÃ
00000020: FF  
```

Below is the string hex output pre_lambda and  captured value  for `Arduino String`

````
Decoding Pre: Cmd: É•rgbwwïÃä+Ã‚, stringlen = 36, actual len = 36
00000000: 83 A5 72 67 62 77 77 95 CC 8A 2B CC E2 00 00 A4  É•rgbwwïÃä+Ã‚..§
00000010: 74 69 6D 65 CD 01 2C AA 62 72 69 67 68 74 6E 65  timeÕ.,™brightne
00000020: 73 73 CC FF                                      ssÃˇ

Decoding Capture: Cmd: É•rgbwwï
00000000: 83 A5 72 67 62 77 77 95 00 00 00 00 00 00 00 00  É•rgbwwï........
00000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000020: 00                                               .

Decoding Cmd: É•rgbwwï
JSON err: NotSupported
```

so you can see that it is truncated and the decode fails. 
I've not got to the bottom of it..  does msgPack allow nulls in the stream and is this causing Arduino string to fail?

bblanchon, 2019-07-15:
Hi @sticilface,

Please apologize for the late answer.

Yes, MessagePack document often contains nulls.
Yes, Arduino's String doesn't support nulls inside the string.
So we cannot use String to store MessagePack document.

I could probably force the compilation to fail, but I'm not sure it's wise because the code is generic, and I fear it could fail for legit usages too.

As mitigation, I added a paragraph in the documentation that warns about this problem.

Best Regards,
Benoit

sticilface, 2019-07-15:
I just saw this and thought that I missed it.  Came here to say that I missed the paragraph in the documentation... again.... then found that you had just added it.. 

thank you again!

closed 2019-07-15 by sticilface

bblanchon, 2019-07-16:
You're welcome.
Thank you for sharing your experience, it will certainly help other users.