bblanchon/ArduinoJson #1047: json buffer problem past 1 week i keep on trying it is showing this problem

sujithkumarjp, 2019-07-14:
Build options changed, rebuilding all
In file included from C:\Users\admin\Documents\Arduino\libraries\firebase-arduino-0.3\src/Firebase.h:30:0,

             from C:\Users\admin\Documents\Arduino\libraries\firebase-arduino-0.3\src/FirebaseArduino.h:22,

             from C:\Users\admin\Documents\Arduino\libraries\firebase-arduino-0.3\examples\FirebaseDemo_ESP8266\FirebaseDemo_ESP8266.ino:21:
C:\Users\admin\Documents\Arduino\libraries\firebase-arduino-0.3\src/FirebaseObject.h:109:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

 std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

       ^
In file included from C:\Users\admin\Documents\Arduino\libraries\firebase-arduino-0.3\src/FirebaseArduino.h:22:0,

             from C:\Users\admin\Documents\Arduino\libraries\firebase-arduino-0.3\examples\FirebaseDemo_ESP8266\FirebaseDemo_ESP8266.ino:21:
C:\Users\admin\Documents\Arduino\libraries\firebase-arduino-0.3\src/Firebase.h:86:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

       ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

bblanchon, 2019-07-16:
Hi @sujithkumarjp,

I think firebase-arduino still uses ArduinoJson v5, so you need to downgrade ArduinoJson from the library manager.

Best Regards,
Benoit

sujithkumarjp, 2019-07-16:
Thank you so much

bblanchon, 2019-07-19:
You're welcome @sujithkumarjp.
Thank you for using ArduinoJson.

closed 2019-07-19 by bblanchon