bblanchon/ArduinoJson #1048: Trouble with String type on Particle device

mhazley, 2019-07-15:
I have a question regarding the use of `ArduinoJson` on my Particle Boron.

Some details:
* Particle Boron is using Device OS `v1.1.1`
* Arduino JSON is `v6.11.2`
* Compiler is `arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1`

I have a class called `Message` with a `DynamicJsonDocument` called `json_buffer` as a private member. 

Message.h:

```
#ifndef MESSAGE_H
#define MESSAGE_H

#include "Particle.h"
#include "ArduinoJson.h"

class Message
{

private:
    DynamicJsonDocument json_buffer;
    JsonArray data;
    int index = 0;

public:
    Message() : json_buffer(1024) {}
    bool add_record( float lat, float lon );
    String serialize_message( void );
    void init( String id );
};

#endif // MESSAGE_H
```

The idea is that it I can instantiate and call `init()` which attempts to set an ID field and then create an array called data into which I will add batches of data records.

This is `init()`:

```
void Message::init(String id)
{
    this->json_buffer["id"] = id;
    this->data = this->json_buffer.createNestedArray("data");
}
```

…but the compiler doesn’t like this (for the line `this->json_buffer["id"] = id;`):  

```
In file included from /Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:13:0,
                 from /Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from /Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from /Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson.hpp:21,
                 from /Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/mhazley/src/cat-609//src/message.h:6,
                 from /Users/mhazley/src/cat-609//src/message.cpp:2:
/Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp: In instantiation of 'typename ArduinoJson6112_11000::enable_if<(! ArduinoJson6112_11000::is_array<TValue>::value), ArduinoJson6112_11000::MemberProxy<TParent, TStringRef>&>::type ArduinoJson6112_11000::MemberProxy<TParent, TStringRef>::operator=(const TValue&) [with TValue = String; TObject = ArduinoJson6112_11000::JsonDocument&; TStringRef = const char*; typename ArduinoJson6112_11000::enable_if<(! ArduinoJson6112_11000::is_array<TValue>::value), ArduinoJson6112_11000::MemberProxy<TParent, TStringRef>&>::type = ArduinoJson6112_11000::MemberProxy<ArduinoJson6112_11000::JsonDocument&, const char*>&]':
/Users/mhazley/src/cat-609//src/message.cpp:8:29:   required from here
/Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:39:5: error: call of overloaded 'set(const String&)' is ambiguous
     getOrAddUpstreamMember().set(src); 
```

If I change the first line of the function to `this->json_buffer["id"] = id.c_str();` everything is fine, as this passes a `const char*`.

I thought maybe I had to be using a json_object so tried this:

```
JsonObject obj = this->json_buffer.to<JsonObject>();
obj["id"] = id;
```
But the same error.

Reading around sort of led me to think that I could use `String` directly with the functions - so am I doing something terribly wrong that I cannot see here? 

As a related aside, I also get a similar problem when trying to serialize... I’ve added the `add_record` code as well to illustrate how I am filling the json array.  

```
bool Message::add_record( float lat, float lon )
{
    bool ret = false;

    if( this->index < 60 )
    {
        JsonObject record = this->data.createNestedObject();

        record["lat"] = lat;
        record["lon"] = lon;

        this->index++;
        ret = true;
    }

    return ret;
}

String Message::serialize_message()
{
    String m;
    serializeJson(this->json_buffer, m);
    return m;
}
```

This generates (for the line `serializeJson(this->json_buffer, m);`):

```
In file included from /Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson.hpp:34:0,
                 from /Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/mhazley/src/cat-609//src/message.h:6,
                 from /Users/mhazley/src/cat-609//src/message.cpp:2:
/Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp: In instantiation of 'size_t ArduinoJson6112_11000::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson6112_11000::BasicJsonDocument<ArduinoJson6112_11000::DefaultAllocator>; TDestination = String; size_t = unsigned int]':
/Users/mhazley/src/cat-609//src/message.cpp:41:39:   required from here
/Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:102:35: error: no matching function for call to 'serialize(const ArduinoJson6112_11000::BasicJsonDocument<ArduinoJson6112_11000::DefaultAllocator>&, String&)'
   return serialize<JsonSerializer>(source, destination);

/Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Serialization/serialize.hpp:53:69: note:   template argument deduction/substitution failed:
/Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Serialization/serialize.hpp: In substitution of 'template<template<class> class TSerializer, class TSource, class TString> typename ArduinoJson6112_11000::enable_if<ArduinoJson6112_11000::IsWriteableString<TString>::value, unsigned int>::type ArduinoJson6112_11000::serialize(const TSource&, TString&) [with TSerializer = ArduinoJson6112_11000::JsonSerializer; TSource = ArduinoJson6112_11000::BasicJsonDocument<ArduinoJson6112_11000::DefaultAllocator>; TString = String]':
/Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:102:35:   required from 'size_t ArduinoJson6112_11000::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson6112_11000::BasicJsonDocument<ArduinoJson6112_11000::DefaultAllocator>; TDestination = String; size_t = unsigned int]'
/Users/mhazley/src/cat-609//src/message.cpp:41:39:   required from here
/Users/mhazley/src/cat-609/lib/ArduinoJson/src/ArduinoJson/Serialization/serialize.hpp:53:69: error: no type named 'type' in 'struct ArduinoJson6112_11000::enable_if<false, unsigned int>'
```
The docs suggest (to me) I can do the above: `serializeJson(const JsonDocument& doc, String& output);` but perhaps I am being stupid here - should I be using `char[]` instead of the String type?  

Any help at getting past these few issues would be greatly appreciated - I have tried diving down into the templated code but getting a bit lost.

bblanchon, 2019-07-16:
Hi @mhazley,

I answered your question in the FAQ:

* [How to use the String class on Particle?](https://arduinojson.org/v6/faq/how-to-use-the-string-class-on-particle/)

Best Regards,
Benoit

mhazley, 2019-07-16:
Thanks @bblanchon - I literally googled for 30 mins on it and didn't find that page. Appreciate the fast response!

closed 2019-07-16 by mhazley

bblanchon, 2019-07-17:
The page didn't exist when you searched; I added it after seeing your question 😉

bugfishteam, 2019-08-11:
Can I just say thank you, you folks are amazing.