bblanchon/ArduinoJson #1050: Using std::stream with StdStreamReader

brettbeeson, 2019-07-16:
I'm using ArduinoJson on an ESP32 with ESP-IDF V3.2.2. I'd like to deserialise a JSON object from a file. 

I need to use standard POSIX file tools, or std::stream. I can't use Arduino SD.  Hence I've tried:

`FILE *file;`
`file = fopen("whatevs.json","r");`
`DeserializationError error = deserializeJson(doc, file);`
and
`std::ifstream file("whatevs.json);`
`ArduinoJson::StdStreamReader ssr(file);`
`DeserializationError error = deserializeJson(doc, ssr);`

Without any luck. It seems (posix) FILE is not supported in deserialize and I can't find any documentation / examples on StdStreamReader.

How can deserialize from a FILE or std::stream? Or maybe I have to read to a String first?

Thanks! Brett

bblanchon, 2019-07-17:
Hi Brett,

[deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) supports `std::istream`, so you should be able to use `std::ifstream`.

The support for `std::istream` is probably disabled, so you need to define [ARDUINOJSON_ENABLE_STD_STREAM](https://arduinojson.org/v6/api/config/enable_std_stream/) to 1.

`StdStreamReader` is an implementation detail that you don't need to care about.

Best Regards,
Benoit

brettbeeson, 2019-07-17:
Hiya,

Okay, I've got that working with the #define. Thanks for your help.

Cheers,

Brett

closed 2019-07-17 by brettbeeson