bblanchon/ArduinoJson #1051: String comparison fails when key value is assigned with `const char *` but not String

sticilface, 2019-07-16:
I've written a function to compare Json objects and bail if the key, values are not found or a different in the target.  The target can have more keys.  

here is the function

```cpp
bool DeviceHandle::canHandle(JsonObject & event)
{
	DynamicJsonDocument handleDoc(256); 
	DeserializationError error = deserializeMsgPack(handleDoc, _handle);   //serializeMsgPack

	if (!error) {
		if (handleDoc.is<JsonObject>()) {
			Serial.print("Comparing handleDoc: ");
			serializeJson(handleDoc, Serial); 
			Serial.print("\nwith event:          ");
			serializeJson(event, Serial); 
			Serial.println(); 
			for (JsonPair kv : handleDoc.as<JsonObject>()) {
				if (!event.containsKey(kv.key())) { 
					Serial.printf("  no match, doesn't contain key: %s\n", kv.key().c_str()); 
					return false;  //  if it doesn't contain the key  
				} else if (event.containsKey(kv.key()) && kv.value() != event.getMember(kv.key())) {
					String H;
					String E;
					serializeJson(kv.value(), H);
					serializeJson( event.getMember(kv.key()) , E);
					Serial.printf("  no match, %s [%s != %s]\n", kv.key().c_str(), H.c_str(), E.c_str() ); 					
					return false;  // 
				}
			}
			Serial.println("Event Matches ** ");
			return true; //  if is contains all keys ==. 
		}
	} else {
		Serial.printf("canHandle: Deserialisation error: %s\n", error.c_str()); 
	}

	return false; 
}
```

Now the input to this function is shown below:  `type` is a string assigned to the key `touchtype`. 

```cpp

template <>
bool PacketToJson<TouchPacket>(const Packet<TouchPacket> & packet, JsonObject & root)
{
	root["type"] = "TouchPacket"; 
	const char * type = nullptr; 
        //String type; 
	switch (packet.data.touchType.c) {
		case Touch_t::BUTTON_SHORT_PRESS : { type = "short" ;   break; }
		case Touch_t::BUTTON_LONG_PRESS  : { type = "long"  ;   break; }
		case Touch_t::BUTTON_HOLD        : { type = "held"  ;   break; }
		case Touch_t::BUTTON_LEDS        : { type = "leds"  ;   break; }
		case Touch_t::READY              : { type = "ready" ;   break; }
		case Touch_t::NONE               : { type = "none"  ;   break; }
	}
	root["touchtype"] = type;
	root["button"] = packet.data.value; 

	return true;
}
```
 I initially used `const char *` but the debug output is shown below, and the line `kv.value() != event.getMember(kv.key())` was evaluating as false, but you can see from the debug that the values are the same. 

```
Comparing handleDoc: {"source":"TouchTest","touchtype":"short","button":1}
with event:          {"source":"TouchTest","pjonID":10,"time":29463,"type":"TouchPacket","touchtype":"short","button":1}
  no match, touchtype ["short" != "short"]  <---   Now this should be true!!!
Comparing handleDoc: {"source":"TouchTest","touchtype":"long","button":1}
with event:          {"source":"TouchTest","pjonID":10,"time":29463,"type":"TouchPacket","touchtype":"short","button":1}
  no match, touchtype ["long" != "short"]
Comparing handleDoc: {"source":"TouchTest","touchtype":"held","button":1}
with event:          {"source":"TouchTest","pjonID":10,"time":29463,"type":"TouchPacket","touchtype":"short","button":1}
  no match, touchtype ["held" != "short"]
```

If I then change `const char *` to String: 

```
[TouchTest] parseEvent: _handles.empty()
Comparing handleDoc: {"source":"TouchTest","touchtype":"short","button":1}
with event:          {"source":"TouchTest","pjonID":10,"time":15645,"type":"TouchPacket","touchtype":"short","button":1}
Event Matches ** 
Comparing handleDoc: {"source":"TouchTest","touchtype":"long","button":1}
with event:          {"source":"TouchTest","pjonID":10,"time":15645,"type":"TouchPacket","touchtype":"short","button":1}
  no match, touchtype ["long" != "short"]
Comparing handleDoc: {"source":"TouchTest","touchtype":"held","button":1}
with event:          {"source":"TouchTest","pjonID":10,"time":15645,"type":"TouchPacket","touchtype":"short","button":1}
  no match, touchtype ["held" != "short"]
```

The output works.  

So it looks like JsonVariant `== operator` is not getting it quite right when comparing strings held in the document with those held as a pointer.  

Any thoughts?

sticilface, 2019-07-16:
I've found another example that doesn't work:
```
Comparing handleDoc: {"source":"TestLED","type":"STATUSpacket"}
with event:          {"source":"TestLED","pjonID":45,"time":1224576,"type":"STATUSpacket","millis":1527186,"error":0,"git":"ea26755","compileT":"Jul  6 2019 15:23:44"}
  no match, type ["STATUSpacket" != "STATUSpacket"]
```

Is is failing here as STATUSpacket != STATUSpacket, but when i serialise the jsonvariant you can see the strings are equal.  The source of these strings is a bit more complex as they are copied from msgpack documents. 

I'd be keen to know more about how the comparators work using jsonVariant.  there is only a single line saying they work but how deep do they compare if it is an array or object?

bblanchon, 2019-07-19:
Hi @sticilface,

Thank you very much for this report, but there is too much code here.
Can you create an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?

Regards,
Benoit

bblanchon, 2019-07-19:
OK, I think I managed to reproduce the bug.

sticilface, 2019-07-19:
Sorry i realised later that it was not really a minimal example...  I'd spent a good couple of hours trying to track down where it had gone wrong and had lost the will after that:)

sticilface, 2019-07-19:
How does `==` operator work when comparing to JsonVariants that are JsonObjects/Arrays.  Does it compare all keys and values?

closed 2019-07-19 by bblanchon

bblanchon, 2019-07-19:
Yes, it compares all keys and values.

commit b54de58e (2019-07-19): Fixed comparison of `JsonVariant` with mixed strings (closes #1051)