bblanchon/ArduinoJson #1055: QUESTION: Most efficient way to navigate (nested) JSON Array as circular lists?

mbariola, 2019-07-19:
Hi there,
my JSON is composed of an array of unknown length. Each element is an array that contains another array of strings of unknown length, a string "explaining" the elements on the inner array, an int represented as string telling me which one is the current selected item in such array.

I need to efficiently navigate the main array as a circular list, and the same for each contained array. Every time I start the sketch, I'll be starting from the first element in the array. I display the "explanatory" string and current selected value on an attached display.
user has 4 buttons: 
A: 1 item forward
B: 1 item back
C: 1 value forward
D: 1 value back
As an example:

`

    [
    
    ["resolution", ["640x320", "128x64", "1920x1080"], "0"], 

    ["warmth", ["cool", "warm"], 0], 

    ["orientation", ["0°", "90°", "180°", "270°", "Mirrored"], "0"], 

    ["input", ["HDMI", "VGA", "DVI"], "0"]

    ]

`
So, sketch starts displaying "resolution: 640x320"
User Input: A ==> "warmth: cool"
User Input: A==> "orientation: 0°"
User Input: D==> "orientation: Mirrored"
User Input: D==> "orientation: 270°" (will be updating the "int" marking current selected value)
User Input: B==> "warmth: cool"
User Input: C==> "warmth: warm" (will be updating the "int" marking current selected value)
User Input: C==> "warmth: cool" (will be updating the "int" marking current selected value)

... and so on.

By reading the docs, I saw only a way to walk the linked list forward, and I seem to understand that the current element(array) in this case need to be walked anyway, to access the next one as size of elements is not cached.

Am I missing something? at any rate the best way to do this "nested" circular management would be greatly appreciated!
Trying this all on an ESP32 via Platformio

bblanchon, 2019-07-20:
Hi @mbariola,

You should use [the iterators](https://arduinojson.org/v6/api/jsonarray/begin_end/) for the outer array, and [the random access](https://arduinojson.org/v6/api/jsonarray/subscript/) for the inner arrays.

ArduinoJson should automatically convert the string to an integer.

Here is a sketch:

```c++
JsonArray outerArray = doc.as<JsonArray>();
for (JsonArray innerArray : outerArray) {
   const char* message = innerArray[0];
   int index = innerArray[2];
   const char* value = innerArray[index];
   // ...
}
```

Best Regards,
Benoit

mbariola, 2019-07-20:
Hi Benoit, thanks for your answer. I am not sure how to do the backwards -
walking though and how to go forward step by step. The for loop seems to me
like going through the array all at once, while I need to step forward once
or backward only once according to user input through the circular list.
Can you please clarify?

bblanchon, 2019-07-21:
Hi @mbariola,

You could use the JsonArray iterator to move one step, but you could not go forward, because there is no backward pointer in ArduinoJson.

I'll give you the same advice I give everyone else in this situation: copy the content of the JSON document to a custom data structure.
Remember that ArduinoJson is a serialization library, it's not a general-purpose container.
If you have special needs for walking the document, convert the JsonArray to something suitable.
Look at [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/) for an example.

Best Regards,
Benoit

mbariola, 2019-07-21:
Thanks @bblanchon! I'll follow your advice. Please feel free to close the
issue
Regards,

MB

bblanchon, 2019-07-22:
You're welcome.
Good luck with your project!

closed 2019-07-22 by bblanchon