bblanchon/ArduinoJson #1058: deserializeJson() from a File stream sometimes gives IncompleteInput error

proddy, 2019-07-22:
I have a similar problem as https://github.com/bblanchon/ArduinoJson/issues/1043 where using `deserializeJson()` with a File stream gives an **IncompleteInput** error.

Now here's the strange part - it did work before I added more functions to the project. The simple test sketch references a rather large class (which has nothing to do with ArduinoJson) and if I add one more function to this class e.g. `void myClass:isAPMode() { \\ do nothing }` the `deserializeJson()` fails with the error each time. Take it out it works again. Weird huh? Doesn't seem to matter what I add to the .cpp file.

The contents of the file is valid and if I load this first into a char array the deserializeJson works fine.

The DATA and PROGRAM is 30% and there is enough heap. Tried with both using `StaticJsonDocument<200> doc` and `DynamicJsonDocument doc(200)`.

Using latest libraries with a Wemos D1 ESP8266 (espressif8266 2.5.2, ArduinoJson 6.11.2) and platformIO 4.0.

I read through the FAQ, GitHub issues and the amazing ebook but can't figure out what is causing it. Any ideas?

proddy, 2019-07-22:
as a work around I'm now reading the file contents first into a char array and then parsing it to deserializeJson(). Still love to figure out though why the size of the project files makes the deserializeJson fail. I may adding a logging proxy and see what it actually pulls in via the stream

bblanchon, 2019-07-23:
Hi @proddy,

Thank you for reporting this puzzling problem.
Also, thank you for researching existing issues and purchasing the book 👍

>  I add one more function to this class [...] fails with the error each time

This sure sounds like stack corruption.
Are you storing large buffers in the stack?
Can I have a look at the source?

Best Regards,
Benoit

proddy, 2019-07-23:
Good point, didn't think of that. I could be blowing the 4KB stack. I guess I could do a quick test in the code to see how much free stack is available before the calls.

All the large buffers in my code are stored in PROGMEM which are used mainly for serving the web pages. Many other libs are used and initialized in the class constructor using 'new' so assume they land on the heap. For example

```
 _webServer    = new AsyncWebServer(80);
 _ws           = new AsyncWebSocket("/ws");
```

I'll see if I can strip down the code to something readable to share, that still causes the error.

proddy, 2019-07-28:
couldn't figure it out so moved on with a work around. I read the whole json file into a char[] buffer first and then call `deserializeJson`, and that seems to work.

closed 2019-07-28 by proddy