bblanchon/ArduinoJson #1059: Converting json response to char[]

mitko9000, 2019-07-22:
After updating the library this line started to fail

>  strcpy(customVar.name, mLastRedJsonObject["key"]);

where customVar is a struct where name field is 

> const char name[20];

Any idea why that is happening when reaching that line the esp8266 is resseting itself

bblanchon, 2019-07-23:
Hi @mitko9000,

`mLastRedJsonObject["key"]` is probably null.
I bet that the `DynamicJsonDocument` is too small.
Unlike `DynamicJsonBuffer`, `DynamicJsonDocument` doesn't grow automatically; you need to pass the right capacity to the constructor.

BTW, your code would be much safer with:

```c++
strlcpy(customVar.name, mLastRedJsonObject["key"] | "", sizeof(customVar.name));
```

Best Regards,
Benoit

mitko9000, 2019-07-25:
Ok this is my code before updating to latest version this thing worked so I have a global variable that I am reusing its the `JsonObject mLastRedJsonObject;`
Also I have a method that validates if certain key exists:
```
boolean checkForKey(const char* key) {
  if (!mLastRedJsonObject.containsKey(key)) {
    debugSerial.printToSerial("wrong json format for request!!!");
    HTTP.send(400, "application/json", "{\"success\" : false}");
    return false;
  }
  return true;
}
```

and that is the part that is parsing the body for each request:
```
void parseRequestBody() {

  request_authorization();

  StaticJsonDocument<200> doc;
  String post_body = HTTP.arg("plain");
  Serial.println(post_body);

  DeserializationError error = deserializeJson(doc, post_body);

  if (error) {
    debugSerial.printToSerial("error in parsin json body");
    HTTP.send(400, "application/json", "{\"success\" : false}");
  }

  mLastRedJsonObject = doc.as<JsonObject>();
}
```
when I increase the json document size from 200 to 400 for example the chip crashes and restart itself even for simple request like `{"test":1234}` when I revert 400 to 200 it parses ok so I assume that the jsonObject is null at this point but I want to be able to parse something like this `{"first":"<20 symbols here>", "second", "<20 symbols here>", "third": "20 symbols here"}` but the json object is always null what I am doing wrong ?

mitko9000, 2019-07-25:
How can I calculate the size of the StaticJsonDocument<100> bytes I need for staticJsonDocument @bblanchon

bblanchon, 2019-07-26:
To calculate the capacity of the `JsonDocument`, use the [ArduinoJson Assitant](https://arduinojson.org/v6/assistant/).
@mitko9000, this is written virtually everywhere on the site, how did you miss it?

mitko9000, 2019-07-26:
@bblanchon thanks for the feedback well my problem is resolved I had to move the StatisJsonDocument to public scope due to the fact I am reusing it, so this can be closed.

Maybe I missed it because I started using the library and reading the migration script but never actually visited the home page, never hurt to ask :D

bblanchon, 2019-07-27:
Cool. Thanks for the update.

closed 2019-07-27 by bblanchon