bblanchon/ArduinoJson #1060: best way to send json as a web socket message in ArduinoJsonv6?

proddy, 2019-07-24:
ESPSyncWebServer has a nice example [here](https://github.com/me-no-dev/ESPAsyncWebServer#direct-access-to-web-socket-message-buffer) of sending json to a web socket based on ArduinoJsonv5:
```
void sendDataWs(AsyncWebSocketClient * client)
{
    DynamicJsonBuffer jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();
    root["a"] = "abc";
        ..... etc ....
    size_t len = root.measureLength();
    AsyncWebSocketMessageBuffer * buffer = ws.makeBuffer(len);
    root.printTo((char *)buffer->get(), len + 1);
    ws.textAll(buffer);
}
```

Question:

What would the correct way to do this in version 6 without creating first another temporary char buffer to hold the serialized message? I don't think `AsyncWebSocketMessageBuffer` has an implementation of `Print` to stream the output too.

bblanchon, 2019-07-25:
Hi @proddy,

I think it's good to pass the serialized JSON document to `AsyncWebSocketClient` because:

1. it solves the ownership problem: `AsyncWebSocketClient` is responsible for freeing the data
2. it frees the `DynamicJsonBuffer` immediately, so it uses less memory (the size of the serialized buffer is significantly lower than the size of the data structure).

However, what I don't like is the double serialization (first with `measureLength()` and then with `printTo()`); we could get much better performance with a fixed-size buffer (plus, it would prevent [heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html)).

Best Regards,
Benoit

proddy, 2019-07-28:
Thanks. So would something like this be the correct way?

```
    DynamicJsonDocument doc(200); // fixed size

    JsonObject          root = doc.to<JsonObject>();
    root["a"] = "abc";
      ... etc ...

    char   buffer[200]; // create temp buffer
    size_t len = serializeJson(root, buffer);  // serialize to buffer

    ws.textAll(buffer, len); // send buffer to web socket

```

bblanchon, 2019-07-29:
Yes. That's what I would do.

closed 2019-08-11 by bblanchon