bblanchon/ArduinoJson #1061: JsonVariant  to (const char*)

MoFanFr, 2019-07-25:
Hello,

I have the following json  {"ON":92}
and the following code :
```
boolean AnalyzeData(const char* key, JsonVariant value, Command * CMD, Sensors *SensorsValues)
{
  Serial.printf("testAnalyzeData");
  Serial.printf("KEY =%s;Value=%s",key,(const char*)value);
...
}
```

I get the message:

```
testAnalyzeDataGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060730  A0      : 0x800e0bc8  A1      : 0x3ffb19f0  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
 A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ffb8688  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060123  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
 
Backtrace: 0x400014fd:0x3ffb19f0 0x400e0bc5:0x3ffb1a00 0x400e80f6:0x3ffb1d10 0x400e812e:0x3ffb1da0 0x400d4cc4:0x3ffb1de0 0x400d313c:0x3ffb1e80 0x400d3332:0x3ffb1ec0 0x400d33ab:0x3ffb1f50 0x400d3774:0x3ffb1f70 0x400d6425:0x3ffb1fb0 0x40085be9:0x3ffb1fd0

```

As you can see, the first printf works but the bug happens at the second printf.

I don't have any error if i use the following Json {"ON":"92"}.
how are the data stored in Jsonvariant ? 
I just switched from an ESP8266 to an ESP32 . With an ESP8266 and a previous version of the library, I didn't have this error.
Why have I got a different behaviour ?
Thank you for your reply and thank you for your awesome library too.
BR,

bblanchon, 2019-07-26:
Hi @MoFanFr,

If you convert a `JsonVariant` that contains an integer to `const char*`, it returns `nullptr`.
If you call `printf("%s", nullptr)`, the program crashes.

Here are several options to solve your problem.

## 1. cast to integer

```c++
Serial.printf("Value=%d", (int)value);
```

## 2. serialize to a temporary buffer

```c++
char tmp[32]
serializeJson(value, tmp);
Serial.printf("Value=%s", tmp);
```

## 3. serialize directly to Serial

```c++
Serial.print("Value=");
serializeJson(value, Serial);
```

## 4. cast to String

```c++
Serial.printf("Value=%s", value.as<String>().c_str());
```

Best Regards,
Benoit

MoFanFr, 2019-07-29:
Ok thank you for your reply.
I will try it tonight.

closed 2019-08-11 by bblanchon