bblanchon/ArduinoJson #1062: Lazy JSON object creation - feature request

vortex314, 2019-07-29:
As any programmer I like to avoid to type  too much characters and have good readability of the code ( with a good signal to noise ratio ).
So I was wondering if there is a way to do this : 

char json[] = "{sensor:gps, time:1351824120,data:[48.756080,2.302038]}";

instead of : 

char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

And parse both successfully in the same JsonDocument

All these escaped quotes don't help much in readability.
I know this is not according to JSON specs, in my case I had to add a lot of JSON config entries.

Or is there another trick to accomplish this ? 
It would resemble a bit HOCON. https://github.com/lightbend/config/blob/master/HOCON.md

bblanchon, 2019-07-29:
Hi @vortex314,

With ArduinoJson 5, it was possible to remove the quotes around key and values.
Now with ArduinoJson 6, quotes are still optional for keys but are required for values.

In both versions, you can replace double quotes with single quotes, so you don't have to escape the character. That gives:

```javascript
{sensor:'gps',time:1351824120,data:[48.756080,2.302038]}
```

See online demo:  https://wandbox.org/permlink/MzVDnox5513MGUGE

Best Regards,
Benoit

vortex314, 2019-07-29:
Benoit,
as usual you give a helpful and swift response ! Thanks for your help.
The proposed approach works for me. 
Lieven

closed 2019-07-29 by vortex314